samphonic/MoppyClassic #89: Floppy does not respond.

Skystriker, 2015-01-26:
Dear SammyIAm,

first I would like to thank you for sharing this awsome creation with the public.

I recently bought an old floppy and had an arduino UNO laying at home.
So I got to work.

I uploaded your code and the led blinks in sync with the melody.
I also wired everything so the driver led lights up if I apply power.

Pin 12&11 are joined.
Pin 17&19 are also joined and connected to ground pin on digital side (4th pin from top)
Pin 18 goes to 2 (3rd from bottom)
Pin 19 goes to 3 (4th from bottom)

If i click on connect the drive clicks once, same with disconnect.
However if I select a MIDI file and play it, the led on the arduino flashes, but the drive only clicks repeatedly every 1-10 seconds.

Thank you for your support.

Skystriker

SammyIAm, 2015-01-26:
Hi there!  It sounds like you've got your STEP and DIR pins swapped.  (Probably my fault, the diagram in the tutorial video is backwards and I tried to fix it with just annotations).  Just switch around 18 and 19 and you should be fine.

MultiDaxio, 2015-01-27:
Hi pal. Have you checked whether your song is not too high pitched? Use SpeedyMIDI to check. Your notes should be in C2 - C5 interval. Notes higher or lower won't be played. Also qudara check the wiring, I kinda know how that can mess up your contraption :p

MultiDaxio, 2015-01-27:
Also, MIDI channels must be the same as the floppies corresponding to their track.

Skystriker, 2015-01-27:
Thank you for your support!
You guys are amazing.

For testing I tried to play the sample songs (they should be fine ;) ).
I switched DIR and STEP pins. And it works just fine.
Yesterday I switched them over and over again and it didn´t work. 

Thank you so much again.

Skystriker, 2015-01-27:
While connecting the rest of my floppies another problem appeared. :(
I bought 6 of them on ebay labled as used but working.

Only the one I had the wires swapped seems to work.
If I connect PIN 11&12 on the other five and apply power they emit one startup sound (got power).
But the front led doesnt light up (none of the floppies has a jumper).
[image]

What could be wrong? Or are they just broken?

MultiDaxio, 2015-01-27:
Are the channels in Moppy Control App enabled? (Are checkboxes for channels 1 - 7 set to positive?

SammyIAm, 2015-01-27:
It could be that the drives you got are hard wired to be drive A or drive B.  Try connecting 13&14 (Drive A Select) instead of 11&12 (Drive B Select).

Skystriker, 2015-01-27:
@MultiDaxio yes they are
@SammyIAm I tried that already, but it didnt help.

SammyIAm, 2015-01-27:
You'll just have to experiment to get it working then.  You've got one drive working so you know your PC and Arduino setup is fine; it's just a matter of figuring out those floppy drives.  Try connecting pairs of pins until the LED turns on-- you'll find it eventually.

Another possibility is that you're counting the pins backward for this drive (i.e. "34" is actually pin 1)

Skystriker, 2015-01-27:
Ok I will try. Thanks a lot

Skystriker, 2015-01-27:
If I connect a wrong pair, is it possible to damage the floppies?

SammyIAm, 2015-01-27:
I can't guarantee that it won't, but it's very unlikely.  I figured out a lot by just running a screwdriver down the pins, and I haven't killed a floppy yet.

(Okay, I *did* kill one floppy when I accidentally drilled a hole through the circuit board... don't do that)

MultiDaxio, 2015-01-27:
I actually fried one myself. Swapped the negative DC with GND DC. Went to
flames instantly. But swapping input/output pins shouldn't damage them, as
long as you don't mess with them while the floppy is running on.

MoppyDoppy, 2015-01-27:
The Even Pins you can change. The odd are all ground and most of them are already connected (the line with the missing pin)
Connect these to ardruino ground. 
The power supply (red+black, not yellow)
Do not connect Ground of IO with Ground of Powersupply

MoppyDoppy, 2015-01-27:
Maybe take away all others and use the drive witch works. Connect this one to the next 2 pins on the ardruino. If it works hardware problem

The test midi should have 8 tracks (I used whichs works on the first and duplicated the track 8 times)

MoppyDoppy, 2015-01-27:
For testing connect a Resistor 330 ohm and behind a led between ground of ardruino and the output pins.
If you do not see lights flashing on one of the pins you have a software problem

Skystriker, 2015-02-19:
Thanks a bunch again. Like I said before I they are from different brands.
By testing with the screwdriver I found the secret:

**You have to have a floppy disk inserted and play around with the pins.
Some of the floppies worked on Pin 11&10 some on 10&9**

It is strange but actually works for me.
I thought this might be helpful.

Thank you again for your amazing support. 
(Maybe you can guess I am in a good mood listening to my floppies :D)

closed 2015-02-19 by Skystriker

SammyIAm, 2015-02-19:
Ah, interesting that you needed to have a disk actually inserted in the drive.  Most drives don't seem to care, but apparently some do.  Thanks for letting us know!