samphonic/MoppyClassic #92: Update Problems

MoppyDoppy, 2015-02-19:
You still use the rxtx Lib. Please have a look in nbproject/project.properties
 run.jvmargs=-Djava.library.path="../SerialDrivers/rxtx-2.1-7-bins-r2/Windows/ch-rxtx-2.2-20081207-win-x64;${env_var:PATH}"
MoppyDesk did not start anymore. Removing the parameter did not work. After copy the old rxrx files back to the serial directory MoppyDesk starts again. (Looks good now)
The new lib NRSerialPort make problems, MoppySim does not start at all.
Exception in thread "AWT-EventQueue-0" java.lang.NoSuchMethodError: gnu.io.RXTXCommDriver.getPortIdentifiers()Ljava/util/Set;
	at gnu.io.NRSerialPort.getAvailableSerialPorts(NRSerialPort.java:116)

I also removed all files and get it all new from github. 
But at all, good work and using make fun.

SammyIAm, 2015-02-19:
Huh, you're absolutely right.  My local repository seems to have the updated project.xml file, but it didn't get synced to GitHub.  It's not in my gitignore file either, but I've been having some weird issues in that regard.  Maybe time to delete and recheckout my local repo.

SammyIAm, 2015-02-19:
Okay, I did an archive of the older version of Moppy, and it now thinks that's a "release" for some reason.  I will have to "release" the new branch I guess.  Will update that sometime today.  Thanks again for finding that.

SammyIAm, 2015-02-19:
Well the good news is: The Download Zip button should work fine now.  Give it another try and let me know how things go.

The bad news is: I have inadvertently deleted the old version of Moppy-- it appears to be gone for good.  A poignant reminder of the transience of life perhaps... less clutter I suppose.  =)

closed 2015-02-19 by SammyIAm

MoppyDoppy, 2015-02-20:
Hi, I got it working (for testing without drives, because I am not at home)

I uses NetBeans and the managed Plattform had still the RXTXComm.jar (due to Installation). I had to to remove the jar file from the lib. The RXTXcomm.jar in lib  and the included in nrjavaserial-3.9.3 version  are not compatible/the same.

SammyIAm, 2015-02-20:
Glad you got it working!  And thanks once again for the heads up on the compatibility issue; NRJavaSerial is actually a fork of RXTX so they have a lot of the same classes and probably conflict.  I suspect a few others will be running into the same issue as they try to update.

Skystriker, 2015-02-23:
You are right Sammy :D
Got the same issue (downloaded the update via ZIP button).
Trying to fix it today.