samphonic/MoppyClassic #93: Can't build a clean checkout

nrother, 2015-02-23:
Hi,

I did a clean checkout and started a build with the default parameters of Netbeans. This is all i got:

```
ant -f /Users/niklas/Documents/git/Moppy/Java/MoppyDesk -Dnb.internal.action.name=run run
init:
Deleting: /Users/niklas/Documents/git/Moppy/Java/MoppyDesk/build/built-jar.properties
deps-jar:
Updating property file: /Users/niklas/Documents/git/Moppy/Java/MoppyDesk/build/built-jar.properties
compile:
run:
java.lang.UnsatisfiedLinkError: no rxtxSerial in java.library.path
	at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1857)
	at java.lang.Runtime.loadLibrary0(Runtime.java:870)
	at java.lang.System.loadLibrary(System.java:1119)
	at gnu.io.RXTXCommDriver.<clinit>(RXTXCommDriver.java:84)
	at gnu.io.NRSerialPort.getAvailableSerialPorts(NRSerialPort.java:115)
	at moppydesk.outputs.MoppyCOMBridge.getAvailableCOMPorts(MoppyCOMBridge.java:108)
	at moppydesk.ui.ChannelOutControl.initComponents(ChannelOutControl.java:133)
	at moppydesk.ui.ChannelOutControl.<init>(ChannelOutControl.java:25)
	at moppydesk.ui.MoppyControlWindow.setupOutputControls(MoppyControlWindow.java:74)
	at moppydesk.ui.MoppyControlWindow.<init>(MoppyControlWindow.java:55)
	at moppydesk.MoppyUI.startup(MoppyUI.java:49)
	at moppydesk.MoppyUI$1.run(MoppyUI.java:87)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:744)
	at java.awt.EventQueue.access$400(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:697)
	at java.awt.EventQueue$3.run(EventQueue.java:691)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:714)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
Exception in thread "AWT-EventQueue-0" no rxtxSerial in java.library.path
	at gnu.io.NRSerialPort.getAvailableSerialPorts(NRSerialPort.java:126)
	at moppydesk.outputs.MoppyCOMBridge.getAvailableCOMPorts(MoppyCOMBridge.java:108)
	at moppydesk.ui.ChannelOutControl.initComponents(ChannelOutControl.java:133)
	at moppydesk.ui.ChannelOutControl.<init>(ChannelOutControl.java:25)
	at moppydesk.ui.MoppyControlWindow.setupOutputControls(MoppyControlWindow.java:74)
	at moppydesk.ui.MoppyControlWindow.<init>(MoppyControlWindow.java:55)
	at moppydesk.MoppyUI.startup(MoppyUI.java:49)
	at moppydesk.MoppyUI$1.run(MoppyUI.java:87)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:744)
	at java.awt.EventQueue.access$400(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:697)
	at java.awt.EventQueue$3.run(EventQueue.java:691)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:714)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
BUILD STOPPED (total time: 7 seconds)
```

Maybe there is any special configuration needed, that is not mentioned in the Readme? I also tried building with IntelliJ, with the same result.

MoppyDoppy, 2015-02-23:
Hi

Remove the rxtx....jar from the jdk lib path. The included rxtx and the original don't like each other. 
The environment must not show it anymore

SammyIAm, 2015-02-23:
What @MoppyDoppy said. :)  For some reason, depending on how the RXTX jar was installed/configured previously, it doesn't always get fully removed when updating to the latest version.

Let me know if you get it working, and I'll update the README to mention something about this issue.

nrother, 2015-02-24:
Ok, I found leftovers of rxtx in `/Library/Java/Extensions/` (I'm on a Mac), after removing them everything is fine.

closed 2015-02-24 by nrother

SammyIAm, 2015-02-24:
Thanks for reporting back; glad it's working for you now.