samphonic/MoppyClassic #95: Lost MIDI In Port functionality due to addition of filterControls

solidsnake745, 2015-03-09:
Looks like with the addition of filter controls we lost the functionality of MIDI In. I'd explain more and stuff, but just had eye surgery and can barely see/stare at a white screen. I trust you'll figure this out on your own. Fix is in MoppyControlWindo, line 231. Change seqControls to currentInputPanel. That way whatever the current panel is, the receiver is set properly to filter controls. Please review, I made the quickest change I thought would fix the problem without analyzing it's impact.
[image]

SammyIAm, 2015-03-09:
I'll try to take a look at this as soon as I can.  (I did fail to check MIDI In after those changes though, so that's my bad)

SammyIAm, 2015-04-10:
Sorry it took so long to get to this; busy times.  Your fix was exactly the solution.  Hope your eye's recovered well.

commit 79d7804c (2015-04-10): Make sure currentInputPanel is connected to filter (fixes #95)
closed 2015-04-10 by SammyIAm