samphonic/MoppyClassic #96: Streamlining bass

sandman27and11, 2015-03-09:
While reading MIDIs, is it possible to discern one track as being a tuba, bass (electric, finger, pick, fretless, slap, synth), or contrabass instrument? If that's possible, why not make in an option that those drives only play low frequency notes?

Perhaps a second 10*12 = 120 note array can be given for those drives to play from that consist only of those low frequency notes? Something like

public static int[] microPeriodsBass = {
30578, 28861, 27242, 25713, 24270, 22909, 21622, 20409, 19263, 18182, 17161, 16198, //0
30578, 28861, 27242, 25713, 24270, 22909, 21622, 20409, 19263, 18182, 17161, 16198, //1
15289, 14436, 13621, 12856, 12135, 11454, 10811, 10205, 9632, 9091, 8581, 8099, //2
15289, 14436, 13621, 12856, 12135, 11454, 10811, 10205, 9632, 9091, 8581, 8099, //3
15289, 14436, 13621, 12856, 12135, 11454, 10811, 10205, 9632, 9091, 8581, 8099, //4
15289, 14436, 13621, 12856, 12135, 11454, 10811, 10205, 9632, 9091, 8581, 8099, //5
15289, 14436, 13621, 12856, 12135, 11454, 10811, 10205, 9632, 9091, 8581, 8099, //6
15289, 14436, 13621, 12856, 12135, 11454, 10811, 10205, 9632, 9091, 8581, 8099, //7
15289, 14436, 13621, 12856, 12135, 11454, 10811, 10205, 9632, 9091, 8581, 8099, //8
15289, 14436, 13621, 12856, 12135, 11454, 10811, 10205, 9632, 9091, 8581, 8099, //9
};

On that train of thought, instead of taking out that valuable 10th track, perhaps only tracks with percussion instruments can be taken out.

SammyIAm, 2015-03-09:
These are all interesting ideas, and I encourage you to definitely try playing around with some of these for your own songs.  The problem with including these in the general code is, in short, that MIDI is *super* flexible.

To start with, each `track` can potentially contain messages for multiple `channels`.  And at any point during the MIDI file, each `channel` can be assigned a new `patch` (instrument, or sound).  So it's difficult to tie down a single track to being any kind of specific instrument.

Additionally, what defines a sound as a "bass" sound is usually the pitch (rather than the other way around).  For example, choosing "flute" as your patch, but playing a C1, will still give you a bass sound (kind of pipe organish).

You could argue that the [general MIDI standards](http://www.midi.org/techspecs/gm1sound.php) assign names to certain patches and those could be grouped by instrument type, but what kind of instrument is a bird tweet?  Furthermore, the same standards that say 106 should sound like a banjo also say that channel 10 will *only* contain percussion, and that the other channels will not.  In practice this isn't the case, of course, which is why it's hard to make any kind of assumptions with MIDI.

closed 2015-03-09 by SammyIAm

sandman27and11, 2015-03-09:
Meh, if there's a track with a bass instrument portion in it, I'd expect it to play low notes in most cases. It's good, though, not to restrict bass instruments from playing high notes for every person who uses software (or a flute playing low notes).

*Cracks knuckles*, Alrighty then; I'll take a whack at this. Sorry to pose a specific suggestion as an "issue"!

SammyIAm, 2015-03-10:
Oh no worries!  I wish there was a better Github area for like, non-issue discussion topics.  For the record, even though I'll probably close "issues" that aren't issues, I don't at all mind using the closed issues for discussion.