samphonic/MoppyClassic #97: The underlying Principle!

omartech, 2015-03-18:
I'm sorry that I haven't yet understood this - but what or how does the stepper motor really make the music? 

I mean... what exactly from the MIDI File gets translated to the floppy drive to mimic the sound?

SammyIAm, 2015-03-18:
No need to apologize =)

So, in order:
1. The Java program reads the MIDI file and determines which [MIDI-note](http://newt.phys.unsw.edu.au/jw/notes.html) should be played.
2. There is an array in MoppyPlayerOutput.java that converts MIDI notes to their period (the period being the number of microseconds between peaks if the note were to be represented as a sine wave; or [inverse frequency](http://www.phy.mtu.edu/~suits/notefreqs.html)).
3. The period data is converted a bit to make it easier for the Arduino to deal with.
4. The Arduino tells the stepper motor to move once per period.

If this happened very slowly (like once per second), you would just hear little clicks from the motor as it moved each second.  The actual sound is generated by various mechanical actions (fiction between the head and the drive chassis, vibration of the drive assembly in response to the force from the motor, &c.).

If this happens very quickly though, the resulting sound can be interpreted as a pitch.  Pretty much the same way a two-stroke engine sounds like *put  put  put  put  put* when it's idling, but has a discernible pitch and sounds like *bvrrrrrrrrrrrrrr* when running at speed.

Hopefully that clears things up a bit.  If anything's still unclear though, feel free to ask!

closed 2015-03-18 by SammyIAm

omartech, 2015-05-07:
Ah man.. you're the nicest... thanks.. makes perfect sense. Very cool ! And excellent way to explain things.