samphonic/MoppyClassic #98: Unsure what to do.

Stucktosomemice, 2015-03-26:
Mar 26, 2015 2:39:04 PM java.util.prefs.WindowsPreferences <init>
WARNING: Could not open/create prefs root node Software\JavaSoft\Prefs at root 0x80000002. Windows RegCreateKeyEx(...) returned error code 5.
WARNING:  RXTX Version mismatch
	Jar version = RXTX-2.2-20081207 Cloudhopper Build rxtx.cloudhopper.net
	native lib Version = RXTX-2.2pre2
Mar 26, 2015 2:39:05 PM org.jdesktop.application.LocalStorage getId
WARNING: unspecified resource Application.id using MoppyUI
Mar 26, 2015 2:39:05 PM org.jdesktop.application.LocalStorage getId
WARNING: unspecified resource Application.vendorId using UnknownApplicationVendor
I have set up my moppy, and i cant see a problem, but every time i run it, i get this error. The moppy window opens, and appears to be working fine, but as soon as i click play it seems to constantly be loading the midi file :sos: :sob:

SammyIAm, 2015-03-26:
Alrighty, two things here to note:
1. Generally speaking, anything recorded as a "WARNING" won't interfere with the operation of the program (as long as people are using the WARNING level in the logs correctly).  In this case, those warnings won't cause any issues.
2. Based on those warnings, it looks like you're using an older outdated version of the code (Moppy no longer uses RXTX, and the vendorId warnings were fixed some time ago).  Download the latest version from Github and give it another try.

Stucktosomemice, 2015-03-26:
I managed to get that side working about 10 minutes after, I really should have mentioned that, i forgot to upload the arduino timer. However, once I got that working, I found the motors were running in the floppy disks, and clearly making that click when changing direction, however I'm finding that its not making the sound that is expected, more of a movement of the motor and thats it. Any idea how to fix this problem?

SammyIAm, 2015-03-26:
Which motor are you talking about?  The big spinny motor that just runs constantly, or the stepper motors attached to the long worm gear?

Are they just clicking occasionally?

Stucktosomemice, 2015-03-26:
They click occasionally, and I'm not really 100% sure, it stops and starts occasionally so i guess the stepper, but its most likely the one that doesn't make the sound.

SammyIAm, 2015-03-26:
If they're just clicking occasionally, you've probably got your STEP and DIR pins switched.  Try switching those wires and see if that helps.

Stucktosomemice, 2015-03-26:
That might be it, ill try in the morning. Thanks for the quick response!

Stucktosomemice, 2015-05-06:
finally been able to try, i cant even get it to click occasionally now, everything i had working seems to have stopped, however, this error has just occurred
May 06, 2015 3:58:21 PM java.util.prefs.WindowsPreferences <init>
WARNING: Could not open/create prefs root node Software\JavaSoft\Prefs at root 0x80000002. Windows RegCreateKeyEx(...) returned error code 5.
WARNING:  RXTX Version mismatch
	Jar version = RXTX-2.2-20081207 Cloudhopper Build rxtx.cloudhopper.net
	native lib Version = RXTX-2.2pre2
May 06, 2015 3:58:22 PM org.jdesktop.application.LocalStorage getId
WARNING: unspecified resource Application.id using MoppyUI
May 06, 2015 3:58:22 PM org.jdesktop.application.LocalStorage getId
WARNING: unspecified resource Application.vendorId using UnknownApplicationVendor
May 06, 2015 3:58:56 PM moppydesk.MoppyMainWindow initializeSequencer
SEVERE: null
gnu.io.PortInUseException: Unknown Application
	at gnu.io.CommPortIdentifier.open(CommPortIdentifier.java:467)
	at moppydesk.MoppyBridge.<init>(MoppyBridge.java:28)
	at moppydesk.MoppySequencer.<init>(MoppySequencer.java:33)
	at moppydesk.MoppyMainWindow.initializeSequencer(MoppyMainWindow.java:69)
	at moppydesk.MoppyMainWindow.connectSequencerPressed(MoppyMainWindow.java:430)
	at moppydesk.MoppyMainWindow.access$700(MoppyMainWindow.java:29)
	at moppydesk.MoppyMainWindow$8.actionPerformed(MoppyMainWindow.java:270)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2346)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:252)
	at java.awt.AWTEventMulticaster.mouseReleased(AWTEventMulticaster.java:289)
	at java.awt.Component.processMouseEvent(Component.java:6525)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6290)
	at java.awt.Container.processEvent(Container.java:2234)
	at java.awt.Component.dispatchEventImpl(Component.java:4881)
	at java.awt.Container.dispatchEventImpl(Container.java:2292)
	at java.awt.Component.dispatchEvent(Component.java:4703)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4898)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4533)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4462)
	at java.awt.Container.dispatchEventImpl(Container.java:2278)
	at java.awt.Window.dispatchEventImpl(Window.java:2750)
	at java.awt.Component.dispatchEvent(Component.java:4703)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:751)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:702)
	at java.awt.EventQueue$3.run(EventQueue.java:696)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.awt.EventQueue$4.run(EventQueue.java:724)
	at java.awt.EventQueue$4.run(EventQueue.java:722)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:721)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

May 06, 2015 3:59:00 PM moppydesk.MoppyMainWindow loadSequenceFile
SEVERE: null
java.lang.NullPointerException
	at moppydesk.MoppyMainWindow.loadSequenceFile(MoppyMainWindow.java:395)
	at moppydesk.MoppyMainWindow.initializeSequencer(MoppyMainWindow.java:80)
	at moppydesk.MoppyMainWindow.connectSequencerPressed(MoppyMainWindow.java:430)
	at moppydesk.MoppyMainWindow.access$700(MoppyMainWindow.java:29)
	at moppydesk.MoppyMainWindow$8.actionPerformed(MoppyMainWindow.java:270)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2346)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
[... 52 lines cut ...]
	at java.awt.Component.dispatchEvent(Component.java:4703)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4898)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4533)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4462)
	at java.awt.Container.dispatchEventImpl(Container.java:2278)
	at java.awt.Window.dispatchEventImpl(Window.java:2750)
	at java.awt.Component.dispatchEvent(Component.java:4703)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:751)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:702)
	at java.awt.EventQueue$3.run(EventQueue.java:696)
	at java.security.AccessController.doPrivileged(Native Method)
[... cut ...]
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:721)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

SammyIAm, 2015-05-06:
Looks like the first error is being caused by a "PortInUseException".  Make sure that you haven't selected any COM ports that are being used by another application, and try again.

QwertzHz, 2015-06-23:
@Stucktosomemice, I think it's important for anyone using Moppy to understand exactly how the drives work. Not to imply you're dumb or anything, but here's a quick explanation of the hardware part:

The stepper motors make the noise. They're 5V and not the big spinny ones. They're in the back (you already know this if you took the covers off). Unlike normal motors, they do not "go" when powered, but rather take "steps" as you toggle the STEP pin.

The floppy's chips take care of the complicated work with the STEP pin and such, so to make the motor take one step, you turn the current to the STEP pin from low to high (and then, presumably, back to low so you can repeat the process).

The DIR pin controls the step direction (obviously). When it receives low current, the stepper motor steps away from the reset position, and towards it when the current is high (correct me if I'm wrong, somebody).

The clicking you're hearing is steps. One click is one step. Ideally they take a high number of steps per second in order to make musical notes. Random clicking likely means the STEP and DIR pins are swapped (STEP is on the right, DIR on the left); it's stepping when the Arduino thinks it's changing the direction. That's why @SammyIAm suggested you swap the pins. As for your software issues, I'm not yet too knowledgeable about the Java side.

Something important to mention is that the grounds of the STEP pins, DIR pins, Arduino, external power, and motor power should all be connected. Varying ground levels due to separation will really make it difficult to do much of anything.

solidsnake745, 2016-03-02:
Closing as there has been no update to this for a while.

closed 2016-03-02 by solidsnake745