samphonic/MoppyClassic #99: Motors get stuck and Midi-Editing Question

themaxx2k, 2015-05-20:
Hi!
Thanks for this piece of damn cool software! Works great. But i have 2 Questions:
1. Sometimes some of the Motors get stuck in End-Position, also reset doesn't bring the rw-head back to start position. Only by manually spinning the motor-spindle it gets unstuck and starts to move again. Is this an issue of my floppies or how could this be fixed?
2. For Midi-editing: does it matter which instrument is selected or whats the best type of instrument to select for the drive-channels?

Yours Markus

SammyIAm, 2015-05-20:
Hi Markus,

Sometimes a floppy will miss an instruction to move the read head, but this doesn't usually cause them to get off by more than a single track, and resetting should fix the issue.  If your heads are physically stuck (which is what it sounds like) it's probably an issue with your particular drive.  You could try to figure out what they're sticking on and remove it / lubricate it.  Alternatively, you could try editing the Arduino code to lower the MAX_POSITION values from 158 to something smaller (even numbers only, read the comment).  This will reverse the direction of the head before it reaches the end which may prevent it from getting stuck.

The MIDI instrument has no effect on the drives.  I usually go with a synth option (like square wave) since it sounds more like the drives will than, say, trombone.  That way when I'm listening to the MIDIs I can get a better idea how it will sound on the drives.

themaxx2k, 2015-05-20:
Will give it a try and report. Thx for your great work. Btw. connecting Harddrive as Wave Output for an channel with a litte AudioAmp (5V 3w) does a great job for high frequencies, and it could be driven by MoppyDesk and a line-out of th PC for one or two channels. Therefor a suitable instrument is required. Will use 4 of them (2 parallel, cause one coil has about 12 Ohms, so down to 6 Ohm which is suitable for the tiny amplifier) as channel 9 and 10. Channel 1 to 8 as com to the floppies. Maybe this is worth an entry in your wiki.

SammyIAm, 2015-05-20:
The beauty of wikis is that anyone can edit them ;)

themaxx2k, 2015-07-27:
Request to everyone: looking for all sort of Floppy-ready Midifiles. Anyone here who wants to share?

solidsnake745, 2016-03-02:
Closing as there has been no update to this for a while.

closed 2016-03-02 by solidsnake745