samphonic/MoppyClassic #100: Crazy idea about drive pooling

tmp6154, 2015-05-26:
You'll probably think I'm crazy, but there used to be a problem, where Moppy Created duplicate transmitters.

Yes, that was clearly a bug, but under certain circumstances it enhanced drive pooling feature. It's very hard to describe, what I mean, but it allowed me to play many MIDI tracks on drives with little editing. Say, somehow it magically corrected note lengths and pooling worked generally much better (when it correctly glitched).

I can give an example. Here's one of MIDI files. The problem here is that it was autoconverted from С64 tune, so it has 1/64 note lengths everywhere. Here it is:

https://drive.google.com/file/d/0BxmVjdIT1OOfM1BWZWNMQW9HUWc/view?usp=sharing

On new versions of Moppy, it results in floppies playing "clicks" (it's only logical, there are only 1/64 notes after all). However, on old versions, when properly glitched, same MIDI file produced great sound, I've recorded one such example:

http://www.youtube.com/watch?v=Eh40u0xCo-0&feature=youtu.be

Surprisingly enough, it sounded as good as manually arranged song for 6 drives. And note lengths were corrected, somehow.

Similar glitch worked great for many other songs, producing way better sound than unglitched pooling. It greatly simplifies MIDI processing, while giving you a nice sound.

I wonder if it's possible to introduce controllable simulation of this glitch, like make transmitter send N messages?

SammyIAm, 2015-05-26:
There definitely was an issue with that; I remember being very confused for a while until I figured out that each note being played like 3 times on top of itself.

It's probably best not to actually duplicate transmitters since that can cause some other issues.  I have a theory about what was causing the notes to work better though, and it might be possible to leverage this:

The way drive pooling handled "note off" events was by looking at all the current notes for one that matched, and then turning the first one it found off.  Because of the duplication issue (and because sometimes the same note was played in another channel), notes seemed to get left on longer than they should have sometimes.  Normally with pooling though, you didn't really notice because pretty quickly a new note came and overwrote the note that was hanging over.

You could probably simulate this then by just blocking all the note-off events from being sent.  Every note would just keep playing until another note from the pooling came and overwrote it.  Alternatively, you might be able to find some sort of MIDI editing macro that would increase all the note lengths (like double or quadruple them).

Life is super busy for me right now, so it's unlikely that I'll get to this soon (I'll leave the issue open though), but if you take a stab at it yourself and have success, definitely let me know.

tmp6154, 2015-05-26:
Hmm, your theory actually makes a lot of sense, since there used to be a lot of drives, stuck on playing notes after song ends. But wouldn't duplication also cause same note to play on N drives with pooling? Not sure if it has any good effect, though. I'll try to check it as soon as I reassemble my floppy setup.

SammyIAm, 2015-05-26:
Duplication would cause the note to play on multiple drives, but only until the first note-off event fired.  So even if multiple drives were playing the notes, they would still sound short and choppy as without duplication were it not for the hold-over issue.

tmp6154, 2015-05-27:
Hmm, I've tried to completely suppress "note offs" from both NoteFilter and DrivePooler, however, that causes all floppies to play notes continuously, while in video they're clearly stopping periodically. Moreover, that video was recorded with stacking strategy, but without note offs stack gets filled instantly and because it's never freed, it just plays same notes until drives are reset. Round robin yields slightly better results, since it cycles through drives, but they still never stop like in video. There must've been some other side effect. It feels like duplication caused note-offs to arrive with some delay, increasing note length as result.

SammyIAm, 2015-05-27:
Hmm, yeah, it makes sense that with stacking all the notes would get stuck on... Offhand I can't think of a good way to emulate a delay in the note-off events.  Nor can I figure out how the duplication would cause a delay (I believe you when you say it happened though-- that duplication thing did weird stuff).

Getting notes to play on multiple drives should be easy enough within the NoteFilter class.  You can just call receiver.send() multiple times; the DrivePooler step will spread those notes out to the next available drives.

You could then also mess around with "breaking" things by, for example, only duplicating note-on events in the NoteFilter, but sending a single note-off.  I'm not sure that would give you what you were looking for, but it might be worth a try.

I'm trying to think of other ways to lengthen notes programatically, but it's tricky because each "note" is actually a pair of note-on note-off events.  Your best bet might be a separate routine that can be run on MIDI files that will modify them by pushing all note-off events back by a set amount, or... something like that.  There's probably a way to create a separate thread that receives note events and then releases them after a certain delay, but I quite figure out how it would work.

tmp6154, 2015-05-28:
Actually, your suggestion (about duplicating receiver.send() in NoteFilter) worked greatly! Some MIDIs work better with two sends, some with three, but it produces exactly same rich sound as old versions with transmitter bug produced. I've ended up duplicating all events, both note on and note off.

SammyIAm, 2015-05-28:
That's... weird... but good!  I'm glad you were able to get something that works pretty well without having to be broken like before.

If you're still in experimenting mood, I wonder if just doubling the tracks in the MIDI file would have a similar effect to doubling the notes, or if there's something specific about the way the pooler and filters work together to create that effect.

I'm closing this issue for now since it's no longer an outstanding "issue".  But feel free to keep commenting and discussing here.

closed 2015-05-28 by SammyIAm

RedFox1177, 2015-05-28:
Maybe someone could introduce a drive pooling option that does this? I would, but I don't know any Java at all.

SammyIAm, 2015-05-28:
Well, like @tmp6154 figured out, it would probably best be added in NoteFilter, rather than DrivePooling.

tmp6154, 2015-05-28:
I've tried it with double and triple tracks. While that also causes song to sound better, it's clearly doesn't work exactly same, as with doubling in NoteFilter. For example, with triple tracks, voice 1 seems to be choppy, while with NoteFilter tripling, voice 1 moves smoothly.

SammyIAm, 2015-05-28:
Hmm, well there are some intricacies then.  It's unlikely that I will get a chance to implement it myself anytime soon.

@RedFox1177 , no time like the present for learning some Java.  ;)

RedFox1177, 2015-05-28:
True that! Maybe over the summer.

tmp6154, 2015-05-28:
I think I could implement this. GUI forms are made in NetBeans, aren't they? Inserting a single spinner and making a for loop in NoteFilter, that would multiply notes N times is quite easy, I might try to put it together tomorrow and make a pull request.

tmp6154, 2015-05-29:
Okay, I've implemented it. I tried to follow code and form design style as close as possible. There's a new spinner in note filter box now, which allows to multiply notes N times. It works nicely. I've made a pull request with feature.

SammyIAm, 2015-05-29:
Looks really nice!  I'll want to do a quick test myself first before pulling it, but everything looks good for now.  Hopefully next week I'll get a chance to take a look.  =)

oherik, 2015-07-26:
Hey tmp6154, I just cloned your repo and to me it seems to work great. I duplicated a couple of midi songs with four channels over my eight floppys, which makes every floppy involved in playing the wonderful music. 

Also, SammyIAm, thanks for creating this program. It's a blast!

tmp6154, 2015-07-26:
Thanks, glad you liked it! Hope it will once become a part of official Moppy. It's surprising, how a seemingly "incorrect behavior" from an old fixed bug could be turned into a feature that produces such wonderful and rich sound.

By the way, SammyIAm, I also thank you for creating Moppy, it's a really wonderful program.

SammyIAm, 2015-07-26:
It's been my pleasure, guys!  Also, I promise I haven't forgotten about
this pull request; I'm in the process of moving to another state though, so
things have been a bit busy.  It's sitting in my inbox though so it'll get
done eventually.

Thanks for the additional feedback though Erik; I won't have to be as
thorough with testing knowing someone else has looked at it as well.