samphonic/MoppyClassic #101: More than 16 drives?

RedFox1177, 2015-05-28:
Would it be possible to have more than 16 drives? Like could you run 2 instances and start them at the same time?

SammyIAm, 2015-05-28:
The number 16 comes from the maximum number of channels available to the MIDI protocol.  Since Moppy uses channels to determine which drive to send the notes to, you're limited to 16 unique drives.  There are some potential ways around this though:
- You could physically wire pairs of drives together.  The signal from the Arduino would go to both and they would just mirror each other.  The drives would obviously have to play the same notes at the same time, but it would be louder.
- If you had an Arduino with enough pins (or shift registers), you could use the Arduino to control a larger number of drives.  You'd still only get 16 different input channels, but you could distribute the notes however you'd like from the Arduino code
- You could re-write Moppy to use something other than channels as the differentiator for which drive to send notes to.  If you're reading from a MIDI file, you could use tracks instead of channels, and then you could have a much larger number of drives.
- You could use something like the drive pooling feature to automatically distribute notes over a large number of drives.  You'd need to modify the Moppy and Arduino code to be able to address more than 16 drives.

SammyIAm, 2015-05-28:
If you *really* want to go above and beyond, you could use dynamic data to determine how many drives the Arduino uses to play each note.  So loud sections use 4 drives for a single note, quiet sections just use 1.  That's getting a bit overcomplicated probably though.

RedFox1177, 2015-05-28:
Thanks! I was just seeing how you would do it as an experiment.

closed 2015-05-28 by RedFox1177

MuffyTarkin, 2016-09-21:
Sam, out of curiosity how would you go about switching Moppy from Channels to Tracks? I have three Arduinos, two connected to 8 separate drives, but I'd love to incorporate the third one for additional things like hard drive coils and larger stepper motors, without having to force them to duplicate existing channels.

I already know how to throw another Arduino into the mix, but when it comes to Java I don't know anything and I'm assuming a multitude of UI changes among other under the hood things would need to occur.

Might be an interesting thing to pursue in some free time.

SammyIAm, 2016-09-22:
Well, theoretically, here's how I'd do it:

The current protocol between the Java code and the Ardunio code specifies one byte for the PIN that corresponds to the channel being used; there's no reason that that necessarily needs to correspond to a particular channel from the source MIDI, so at the end of the day what you need is a way to map MIDI events to a particular Arduino/PIN combination.

Because there's a huge variety of channel/track combinations possible (I'm not sure there's a limit on tracks-per-channel) I would adjust the UI to allow adding an arbitrary number of outputs on the right side (i.e. there'd be a + button at the bottom and a - button next to each) and each output would have:
* a dropdown for channel
* a dropdown for track
* the same dropdown for COM port
* a dropdown for PIN number (Only even numbers to work with the current Arduino code)

Each time a MIDI event came out of the sequencer, the Java program would determine which output it matched based on the channel/track, and send the message along on the appropriate COM port with the appropriate PIN number.

You'd mostly have to either bypass or rewrite all the pooling/filtering/adjusting code in the middle of the pipeline, but otherwise it should be fairly doable.