samphonic/MoppyClassic #103: Drive 1 doesn't change directions

Ivoah, 2015-06-09:
I have two drives hooked up to my Arduino, and drive 2 works fine, but drive 1 doesn't change direction. It (almost) works if I manually connect the direction pin to ground and off, but not when plugged into the Arduino. I have everything wired up correctly.

SammyIAm, 2015-06-09:
Hello!  If the direction changes when you manually connect the wire to ground, but not when it's connected to the Arduino, make sure your Arduino's GND pin is well-connected to at least one of the odd-numbered pins on the floppy drive.

Ivoah, 2015-06-10:
It works when I swap the drives, and when I run it with my calculator, but not when run by the Arduino in slot 1

SammyIAm, 2015-06-10:
What do you mean by "swap the drives"?  Is it just one particular drive that's acting up?

And when you say "slot 1", are you referring to pin 1 on the Arduino, or the first drive position (i.e. pins 2 and 4)?

Ivoah, 2015-06-10:
It's only one particular drive that's acting up, and "slot 1" refers to the first drive position (pins 2 and 3)

SammyIAm, 2015-06-10:
Ahhh, okay.  Well if it's just the one particular drive, there might just be something wrong with it.  It's unusual that is seems to work when you manually connect the wires though; perhaps some connection on the drive isn't very good.  It still sounds like there might be some sort of grounding issue between the Arduino and the drives, but otherwise I'm sort of out of ideas.

I'm going to close this issue because it doesn't appear to be software-related, but you're welcome to continue discussion on this thread.

closed 2015-06-10 by SammyIAm

Ivoah, 2015-06-10:
Hmm, well it works when connected to the 2nd drive position (3 and 4) and also when playing music from my calculator (http://www.cemetech.net/projects/item.php?id=38)

SammyIAm, 2015-06-10:
It's possible that that was a typo, but just to confirm: the 2nd drive position is pins **4 & 5**, the 1st position should be pins **2 & 3**.

If your "second" drive is working on 3 & 4, you've got the STEP and DIR wires crossed as well.  (STEP pins should be connected to even numbered Arduino pins, and DIR pins should be on the odd pins)

Ivoah, 2015-06-10:
Yeah, that was a typo, oops