samphonic/MoppyClassic #104: NetBeans doesn't recognize Moppy as a project

RedFox1177, 2015-06-20:
I have recently downloaded moppy, and when i went to start it, NetBeans would not find the project. Any suggestions?

Thanks,
RedFox1177

QwertzHz, 2015-06-22:
NetBeans should recognize the folder Moppy-moppy-advanced/Java/MoppyDesk as a project.

In short, it's looking for a folder, not a file, to represent the project.

SammyIAm, 2015-06-22:
Like @QwertzHz said, you should be able to go to File>Open and then open the MoppyDesk folder.  If that's not working, can you show a screenshot of your Open dialog when you're trying to open the project?

RedFox1177, 2015-06-23:
I will when I get to it. Thanks for your imput!

solidsnake745, 2016-03-02:
Closing as there has been no update to this for a while.

closed 2016-03-02 by solidsnake745