samphonic/MoppyClassic #105: Consistent failure to correctly change drive direction

QwertzHz, 2015-06-22:
I have eight drives connected. The step and direction pins' GND counterparts are grounded as they should be. The ground from the breadboard barrel jack connector (which receives 5V 2A from the wall) is connected to the Arduino's ground. In short, grounds all over the place.

Everything works wonderfully for the most part, with the exception being my third and fifth drives, which tend to lose track of themselves at specific parts in certain MIDI files. While I'm not sure what exactly makes these parts special, one example is about 19 seconds into MrSolidSnake745's "Carol of the Bells", for which the MIDI can be found here: http://puu.sh/iyMXS/62d3fad2c5.mid
I also recorded a short 30-second video of the third drive: https://drive.google.com/file/d/0Bx_OBo4MbVdQSnM5UXBHNktuaTQ

In this example, the drive responsible for channel 3 locks up at the bottom (reset position) and seems to move several steps out before reversing direction and heading back to the bottom. When it hits the bottom, the Arduino clearly doesn't think it has, since it rams itself back and forth like this for a few seconds, using only between a quarter and a half of the full range of motion it usually does.

The drive on channel 5 does a very similar thing, but instead of locking up at the bottom, it locks up in the middle, at neither end. It locks up even more tightly than drive 3, jittering within under a quarter of its range. The area within which it does so shifts somewhat throughout the next bit of the song.

I should note that these things happen in many songs, whether I Constrain Notes or not, and so far only to those two drives. This probably means it's a hardware issue, though either way something's wrong.

I'd appreciate any light shed on both the reason for this and the solution. Thanks!

SammyIAm, 2015-06-22:
Hello there!  First, thanks for the video; that makes it so much easier to see what's going on than trying to have it described with words.

Because the Arduino can't actually sense where the head is (that would be nice though), it's just counting the number of times it's told the head to step, and when it hits about 79 (there are 80 tracks on a high density floppy disk) it reverses direction.

The reason drives get out of sync with the Arduino and crash into things is because some of the signals to step either don't make it to the drive (wiring issues; probably not your problem) or the drive didn't see all of them (either grounding issues, *or* the pulse was too short).

I think probably what you're experiencing is that some of the notes in your MIDI file are too high for that particular drive to play; the pulses between steps are too short and the drive either isn't seeing them, or can't step the head fast enough to keep up.  The top frequency of drives seems to vary a bit by manufacturer, and I tried to choose a reasonably safe upper limit (for the software's constraints), but still sometimes the drives miss some steps.

If you have access to MIDI editing software, try taking the troublesome part of the song down an octave and see if it performs better.  Or try switching out the troublesome drive to a slot that pays more bass parts.

QwertzHz, 2015-06-22:
Thanks.

It would be nice, though probably complicated, to implement a sort of
calibrator. Have each drive, one by one, play higher and higher notes until
they screw up, then record that and set that as the permanent Constrain
Notes roof. Of course, that'd also mean Constrain Notes would have to be
independent of drive. Oh well.

SammyIAm, 2015-06-22:
Yeah, like you said, it would get a bit complicated if each drive needed to have its own constraints.  (There's nothing stopping you from hard-coding some for your drives though).

I'm closing this since it's not a software "issue", but feel free to continue the discussion here if needed.

closed 2015-06-22 by SammyIAm

QwertzHz, 2015-06-23:
I might consider it. I'm fluent in Java, but I'm not 100% familiar with all of the objects and methods used in MoppyDesk. I'm thinking it's closer to 5%, haha.

QwertzHz, 2015-06-23:
Not that I'm pretending to know all about how your program works, but I wonder if somehow there's a miscommunication happening. When I dropped HIGHEST_NOTE in NoteFilter.java from 71 to 51, a drive that never had issues began to do the same thing that 3 and 5 were doing. This strikes me as odd, since I was _lowering_ the ceiling, which, if anything, should make the drives more capable. Any idea what would cause that?

QwertzHz, 2015-06-23:
I now realize I didn't mention that my Arduino is a Mega 2560. Probably makes a difference.

SammyIAm, 2015-06-23:
It does seem unlikely that lowering the ceiling would cause more issues to arise, but I also can't think of any reason that the software would cause that to happen, unless it's either a coincidence, or (very unusually) the drives have a middle-range in which they don't perform well.  I'm not saying it's not possible that there's some communication issues, but I just can't think of where they would arise.

Arduino model shouldn't make any difference that I'm aware of; I'm not using any advanced capabilities of the board, so they should all be pretty compatible.

Can you identify which specific range of notes seems to cause issues for the drives?  When you lowered the ceiling, did 3 and 5 stop having issues?

QwertzHz, 2015-06-23:
I'll do some more extensive testing in a bit. Also, I'm curious: where in MoppyDesk is the info sent to the Arduino?

SammyIAm, 2015-06-23:
The last class that the messages pass through is MoppyCOMBridge.java .  MoppyPlayerOutput.java might be more interesting to you though since it's got more MIDI-note stuff.