samphonic/MoppyClassic #106: Noisy motors

in03, 2015-07-14:
Hey, fellas. Not sure if this is the right place to ask, as I'm not sure if this is an issue with Moppy. All three floppy drives I've tested have had an awful buzzing noise, and I'd thought maybe the head was loose, but the motor itself is causing it. I'd noticed that the travel of the head along the worm drive is not nearly as much as I've seen in other's videos. Am I doing something wrong? They're all different aged Panasonic drives. I've tested dirty earthing and different power supplies. Not sure if this is a common problem or not?

Awesome piece of work, Sammy!
Also, how would hard drives be rigged?

SammyIAm, 2015-07-15:
Is there any way you could post a short video of what you're describing?  It's a little hard to diagnose without seeing it in person...  Some ideas though:
- Could just be those particular drives are noisy / loose.
- If the head isn't moving like it should, it could be a signal issue, or the notes you're trying to play might be too high or low.
- Make sure the drives are grounded to the same point as the Arduino (probably wouldn't cause what you're experiencing, but worth checking)

Thanks! :smile: 

Harddrives use a coil of wire in a magnetic field to control the read head; basically the same way a speaker has a coil of wire in a magnetic field.  All you need to do is connect an audio signal to the coil, and voila: harddrive speaker.

in03, 2015-07-15:
Thanks! For sure! I'll upload a video as soon as I get a chance. 
I've tried a bunch of midi files that MrSolidSnake has used for some of his videos, so I think the notes should be in good range. And I checked that the drives and Arduino were grounded together before I posted. The power-supply is definitely delivering 5 volts too. If it's the model of drive I'm using, is there a brand/model that is known to work well? I have access to a bunch of 5.25" and 3.5" floppy drives, but they're $5 each :wink:  

Awesome! I'd never have guessed that about hard drives. 
Thanks for the prompt reply, sir! I'll get that video up soon :smile:

SammyIAm, 2015-07-15:
I don't know if there's a particular brand or model that works well, but I do know that different models produce different sounds (which is not terribly helpful, I know =P ), so you might try getting just a single one of the other kind of drive and trying it out.

in03, 2015-07-19:
I'll definitely consider getting another one for testing! Sorry for the delay, I've been a busy beaver. Here's a clip of what I'm experiencing. The music playing is meant to be Rick Astley's "Never Gonna Give You Up", but it's kinda buried underneath all the buzzing.
https://www.youtube.com/watch?v=PgOmbeV44HE

I assume it should sound a bit more like this, as this is the same midi file. Though I'm one floppy less.
https://www.youtube.com/watch?v=OinY_Xr2zHg

QwertzHz, 2015-07-19:
Um...yeah. The things that should be moving aren't. At all. I bet your
problem is with the direction pin, or the step counter in the Arduino code.

SammyIAm, 2015-07-19:
Huh, okay, the video definitely helps understand what's happening... but I'm still not sure what's wrong.  =P

You can kind of hear the tune buried in there, and you can see that the drives *do* occasionally reverse directions, so it seems like your setup is pretty much set up correctly.  It looks like each "pulse" of the stepper motor isn't long or strong enough to actually make it take the full step; so instead of stepping forward and backward, it's just sort of twitching in place.

I don't know for sure how to help you, but I have a few ideas:
- If you can get another floppy drive from a different source and give it a try, maybe these two are just sad floppies.
- If you can take the black read-head off of the stepper motor and see if it works any better, maybe there's too much friction/load on the motor to move that quickly.
- Try a different power supply?  Maybe there's not enough power getting to the motors?  (Seems unlikely)
- Try making a MIDI file that just has long, low notes (like C1) in it and see if they behave more normally.  Maybe these drives just don't do well with higher notes.

I wish I had something concrete to tell you, but I can tell you it's closely related to the drives themselves / the stepper motors (and your wiring and software is set up technically correctly).

MoppyDoppy, 2015-07-19:
I had a similar problem when I connect the power supply not only to the left power connection of the floppy.
Only use the ground (GND after pin 13) of the ardruino for the right side of the floppy. Do not connect anything to power (opposite side)

in03, 2015-07-23:
Thanks for the replies, guys. Sammy - I've taken the read head off before and tested with just the motor standing free, not even connected to the rest of the floppy, but it still buzzes and I've tried different power supplies too, but I'll definitely have to give it a midi file with low notes and then try a different drive and I'll let you know if that solves anything :)

MoppyDoppy - Thanks! I'll double check my wiring. So you say not to common ground the power supply to the arduino?

MoppyDoppy, 2015-07-24:
The arduino gets the power from the pc. It does not need more. You can see the wirering of the arduino uno and mega in my YouTube videos (MoppyDoppy) with 8 or 16 drives. 
Or as link
https://m.youtube.com/channel/UCVupT30jd-n1i7CqCu0Zo0g

SammyIAm, 2015-07-24:
Common ground from the power supply to the Arduino should be fine (I mean, ostensibly if you ground anything from the floppy drive to the Arduino, it should be essentially the same thing).  You probably don't want to connect the +5v line from the power supply to power your Arduino while it's connected to your computer via USB though (I've done it once or twice accidentally and nothing bad happened, but it's probably not great for your computer)

MoppyDoppy, 2015-07-25:
Ok. Both of my ardruino uno R3 have problems if common ground from the power supply is connected. I think it makes a difference. But maybe I am wrong or only a few arduino have this problem.

solidsnake745, 2015-08-28:
My 2 cents here. In the past I've had similar problems and I usually blame it on common ground issues. You have to remember that if the ground levels between the floppy drive and your micro controller differ in the slightest it can cause issues with the floppy drives controller board recognizing the rising/falling edge of your signals (both step and direction). That can cause both signals to be misinterpreted. From your videos I'd say that is what is going on here. Something like 50% of the signal changes are interpreted correctly, other half are not.

What I've done is to wire one of the grounds on two of my drives back to my micro controller. Two drives is just for insurance, one should be fine. Since the bottom row of pins on the drives connector is all grounds, I would wire one of those to the Arduino's ground. This should ensure they match and hopefully clear up your issue.

I don't have any real proof of any of the above, it's just an idea, but it's a really simple test. Hope it helps.

QwertzHz, 2015-08-28:
I agree with solidsnake745. Ground can definitely screw it up badly, and it
looks like that's the issue.

yale2a1, 2015-10-21:
@SammyIAm (sorry.. I'm not well engilsh) I don't understand.. My floopy drive just **clacking..** (like Hard drive noisy) what's wrong?
>SammyIAm commented on 25 Jul
Common ground from the power supply to the Arduino should be fine (I mean, ostensibly if you ground anything from the floppy drive to the Arduino, it should be essentially the same thing). You probably don't want to connect the +5v line from the power supply to power your Arduino while it's connected to your computer via USB though (I've done it once or twice accidentally and nothing bad happened, but it's probably not great for your computer)

It's means **'ATX(PC) power supply +5v'** connent Floppy drive? or **don't connect +5v** and connect by arduino +5v?

jazzbassNick, 2015-10-21:
There's technically two ground (earth) connections. PC power supply (the
+5v & ground) to the power side of the floppy. The ground for the motor
(the bottom row of pins) to the ground on the Uno. I daisy chain my bottom
pins (one drive to the next) then the last one goes to the Uno.

Although 'clacking' sounds more like reversed step/direction to me.

I'm sure someone else will chime in to help.

SammyIAm, 2015-10-22:
Do connect the +5v from the power supply to the floppy drive.

Not sure that will solve your clacking though.  Try switching the STEP and DIR pins.

solidsnake745, 2016-03-02:
Closing as there has been no update to this for a while.

closed 2016-03-02 by solidsnake745