samphonic/MoppyClassic #107: Need some help on the hardware part

aFerret, 2015-08-07:
Hello,
A few days ago i finally get the software running, everything fine (Even my Arduino lights on the Rx LED when the midi file plays something), but my floppy drive doesn't move.
I hooked the pins on my protoboard, put the direction and the step wires on pin 2 & 3, but nothing happens.
When i touch the wires manually, the motor moves just a little bit, like in the video (Minute 3:50)
I tried everything, put the direction and step wires invested, put them on pin 4 & 5, even analog pins, and i get nothing.
¿What may be wrong?

I apologize for my bad redaction, but my native language is spanish.

Thanks for everything!

themaxx2k, 2015-08-07:
Did you check the ground connection? Can you manualy check with multimeter if signals are applied to the output pins when midi file is played? Could you make a Photo of your setup?

QwertzHz, 2015-08-07:
I agree with themaxx2k. Ground could be your issue. Wall ground should be
connected to Arduino ground, and both should connect to the floppies'
ground. Uniform everywhere.

aFerret, 2015-08-07:
Wew, i feel so dumb right now.
I accidently put the green and blue stepped wires on to the actual voltage (of the floppy), and the green and blue wires in ground pins.
This is so embarrasing, but i finally made it!. Music flows through them! :)

themaxx2k, 2015-08-08:
Great. One more weird music device is alive :-)

QwertzHz, 2015-08-08:
aFerret, are you color-coding your wires? Doesn't really matter how you do
it but it's very helpful.

SammyIAm, 2015-08-08:
Glad you got it working!  Sorry for being AFK for a bit; I've been without Internet for a few days.

Thanks, @themaxx2k , and @QwertzHz for helping out.

closed 2015-08-08 by SammyIAm