samphonic/MoppyClassic #109: MoppyDesk takes long time to start

orzm, 2015-08-20:
I have the following problem:
If I press "Run" in NetBeans, it takes a long time (10 minutes or more) till the software starts.

My specs:
NetBeans 8.0.2 with JDK 8 64bit at Windows 8.1.1 64bit
i7-4702MQ
8GB RAM

(Sorry for my bad English.)

SammyIAm, 2015-08-21:
Whoa!  That *is* a long time to start!  My first guess was going to be serial-port related issues, but since you also said it freezes on the COM port drop-down, I'm reasonable more sure that that's it.

Could you maybe paste the output from the NetBeans output view that you get when you run it, and indicate where the long pause happens?

I'm not sure there's going to be a lot I can do to help since this seems specific to your system.  I would suggest maybe giving it a try with Java 7 though just to see if that makes a difference.  You could also try grabbing a more recent version of [NRJavaSerial](https://github.com/NeuronRobotics/nrjavaserial) from their GitHub to see if that resolves your issue.

orzm, 2015-08-21:
Thanks for your effort. I tried it on Linux Mint 17 meanwhile, there it runs perfectly. :)

solidsnake745, 2015-08-28:
I think this may be my bad. I'm going out on a limb here and guessing, but there should be an easy test.

This is probably due to my change to the ChannelOutControl. I noticed an issue where if you connected/disconnected a COM device after Moppy had started up, there was no way to update the dropdown to reflect that. So I coded in a refresh function available COM devices and tied it to when the dropdown becomes visible. It's possible it's running this on every ChannelOutControl instance on startup.

Not usually a big deal, however in past experiences I noticed Bluetooth devices will cause the refresh process to take considerably longer. So it's possible, in combination with the above, that this is causing Moppy to take forever to startup for you. It may not necessarily be a Bluetooth devices, but something else that's is taking a while to recognize.

Try commenting out refreshAvailablePorts(); in comComboBoxPopupMenuWillBecomeVisible on the ChannelOutControl class and see if that helps. If so I'll try to code in a more centralized list of COM devices and a manual trigger (a button I suppose) for the refresh.

AlexandruCristian, 2016-01-04:
I have this problem when I run the MoppyDrive and I relly need help!!

WARNING: Could not open/create prefs root node Software\JavaSoft\Prefs at root 0x80000002. Windows RegCreateKeyEx(...) returned error code 5.
java.lang.UnsatisfiedLinkError: C:\Program Files\Java\jdk1.8.0_05\jre\bin\rxtxSerial.dll: Can't load IA 32-bit .dll on a AMD 64-bit platform thrown while loading gnu.io.RXTXCommDriver
Exception in thread "AWT-EventQueue-0" java.lang.UnsatisfiedLinkError: C:\Program Files\Java\jdk1.8.0_05\jre\bin\rxtxSerial.dll: Can't load IA 32-bit .dll on a AMD 64-bit platform

thanks for helping me!!

solidsnake745, 2016-03-02:
@AlexandruCristian, are you still having trouble with this? Just seeing it now, but that error is being caused by using the wrong type of rxtxSerial.dll. Seems you have a 32 bit version when your operating system is 64 bit. Which is odd because I thought Moppy was updated to no longer use the rxtxSerial.dll and instead use a different library. Let me know if you're still having issues.

solidsnake745, 2016-05-09:
Closing as there's been no response for over two months. If you're still having this problem, please create a new issue.

closed 2016-05-09 by solidsnake745