samphonic/MoppyClassic #111: Double check defined note micro-periods

solidsnake745, 2015-09-20:
I've working on some stuff and needed the full range of note periods to try some stuff out. However in doing so I think I possibly found a bug with the periods, but I'm not entirely sure.

The micro-periods defined in MoppyPlayerOutput have a comment for each of the populated rows explaining which octave the notes belong to. The **third** row states it's the **second** octave (0 indexed). 

[image]

I'd always just thought this was just human error when writing the comments, but when I calculated the periods myself I noticed it was off by one octave. So we've possibly been playing drives one octave lower than intended? Is that per design and intentional? Or are there other systems that define notes differently?

For my calculations I referenced two Wikipedia pages:
https://en.wikipedia.org/wiki/Scientific_pitch_notation
https://en.wikipedia.org/wiki/Piano_key_frequencies

With that I came up with the following spreadsheet for the micro-periods: https://docs.google.com/spreadsheets/d/1jNi_arrhXfX19zv5hv4yki_hOsP1rVKY8VlNhPOH6L0/edit?usp=sharing

If my spreadsheet is accurate, here is the updated array.

```java
public static int[] microPeriods = {
        0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
        30578, 28861, 27242, 25713, 24270, 22909, 21622, 20409, 19263, 18182, 17161, 16198, //C1 - B1
        15289, 14436, 13621, 12856, 12135, 11454, 10811, 10205, 9632, 9091, 8581, 8099, //C2 - B2
        7645, 7218, 6811, 6428, 6068, 5727, 5406, 5103, 4816, 4546, 4291, 4050, //C3 - B3
        3823, 3609, 3406, 3214, 3034, 2864, 2703, 2552, 2408, 2273, 2146, 2025, //C4 - B4
        0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
        0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
        0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
        0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0
    };
```

Thank you.

QwertzHz, 2015-09-20:
You know...that could be an actual issue. At one point a couple of months
ago, I set up my floppy system again, and certain drives would break on
certain songs. I didn't understand why, since about half a year ago when I
initially set it up, they worked fine. Maybe they were trying to play notes
that were too high.

solidsnake745, 2015-09-20:
But were they were still trying to play notes? If the notes were too high, then they wouldn't play anything at all since there are 0's at those positions.

The "problem" this would cause is just the wrong octave note being played. So for instance if you wanted to play C2, C1 is actually played. The notes defined are one octave lower than the position they are in is what I'm pointing out.

Also as far as I can remember, this array has never been changed since the conception of Moppy. Looking at the regular Moppy branch, it's the same there also. So I don't think this is the cause for the problem you're describing. Oh it seems Sam must have cleaned out that branch from his main repo. I still have it in my own fork though: https://github.com/solidsnake745/Moppy/blob/master/Java/MoppyDesk/src/moppydesk/MoppyPlayer.java.

QwertzHz, 2015-09-20:
Huh. OK. Guess I still don't know why.

SammyIAm, 2015-10-06:
Hello there!  I'm not dead, really!  I'm so sorry I've been away so much lately; I really need to try to keep up with the GitHub/YouTube/Email messages.

The indexes of the periods are designed to correspond to the MIDI note number.  That is, an A440 is MIDI note number 69.  This is *very much* the standard, and I've been unable to find any examples of that not being the case.  The array of microPeriods is absolutely correct (within a few microseconds here or there due to rounding errors).

The numbering of octaves, on the other hand, is not so concrete.  Based on the naming scheme I chose (I've been using [this page](http://www.phy.mtu.edu/~suits/notefreqs.html) as a reference), that first row would indeed be called C-1.

The important thing in this case is the connection between MIDI note number (e.g. 69) and frequency (e.g. 440Hz), rather than anything to do with the arbitrary octave numbering systems.

closed 2015-10-06 by SammyIAm