samphonic/MoppyClassic #112: initialise_event_info_struct: initialise failed!

GTP95, 2015-09-24:
I've downloaded and compiled your program using Netbeans. I've selected only one channel because I have only one floppy drive connected to my Arduino and loaded a MIDI file from your sample, but when I click on the "connect" button I get this error in Netbeans and your program crashes: initialise_event_info_struct: initialise failed!
I noticed that if I use the clean and build option I get this compilation warnings:
Note: /home/giacomo/Arduino/floppy_music/Moppy-moppy-advanced/Java/MoppyDesk/src/moppydesk/ui/PlaylistControls.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
but I don't think they have something to do with the issue. I didn't installed the "rxtx" replacement you mentioned in the readme file because seems to me that you've included it in the project's files and I never had a previous version of Moppy so I don't have conflicts between rxtx and the new library because I never downloaded rxtx. Can you please help me figuring what's went wrong?

SammyIAm, 2015-09-25:
The error you're getting is definitely serial-connection related.  As you said, the project should include the NRJavaSerial library that you need.  What OS are you trying to run this on (and is it 64 or 32 bit)?

I may not be able to help you too much since the serial software isn't mine.  Could you paste the whole log you get when you try to run things via NetBeans though?  Maybe there's something else useful in there.

GTP95, 2015-09-25:
I was about to do a copy&paste of the log when I realized I couldn't select the right serial interface anymore (the list was empty), so I opened Arduino IDE and it had the same problem (in this case the menu was disabled). I purged Arduino's packets (arduino and arduino-core) and reinstalled them, then I used the option in Arduino IDE's menu to import the TimerOne library (instead of manually copying files as stated the tutorial I was following). I compiled the sketch and uploaded it to Arduino, then I ran Moppy Control Application. Now I can Again select the serial interface (actually automatically selected since it's the only one avaiable) but the initial problem remains. I use Lubuntu 15.04 32 bit and here's the execution log:
ant -f /home/giacomo/Arduino/floppy_music/Moppy-moppy-advanced/Java/MoppyDesk -Dnb.internal.action.name=run run
init:
Deleting: /home/giacomo/Arduino/floppy_music/Moppy-moppy-advanced/Java/MoppyDesk/build/built-jar.properties
deps-jar:
Updating property file: /home/giacomo/Arduino/floppy_music/Moppy-moppy-advanced/Java/MoppyDesk/build/built-jar.properties
compile:
run:
Loaded sequence with 3 MIDI tracks.
initialise_event_info_struct: initialise failed!

If it's OS-related, can you please tell me on wich OS(s) it should run?

GTP95, 2015-09-25:
PS: if I didn't connect the right floppy pins to Arduino the software should run anyway, right?

GTP95, 2015-09-30:
Any idea?

QwertzHz, 2015-10-01:
The software has no idea what's connected. IIRC the program takes no input
at all from the pins.

SammyIAm, 2015-10-06:
Sorry about the delay; I have been irresponsibly absent from GitHub and YouTube lately.

The software should run pretty seamlessly on Windows 7 64bit.  I haven't done any testing myself with any other OSes, but there's no reason it can't be made to work on anything that runs Java.

As @QwertzHz said, the pins are only used for output and shouldn't affect anything (you shouldn't have anything connected to pin 0 or pin 1 though; they're shared with the serial communications and can mess things up a bit there).

It's definitely a serial communications issue.  You may need to build your own NRJavaSerial libraries following the [directions here](https://github.com/NeuronRobotics/nrjavaserial).  And I hate to tell you, but the best solution might be to actually use a different Java Serial library entirely.  You'd have to change a little bit of code in MoppyDesk, but it should be minimal changes.

GTP95, 2015-10-11:
Tanks, I'll do some experiments when I have time.

solidsnake745, 2016-03-02:
Closing as there has been no update to this for a while.

closed 2016-03-02 by solidsnake745