samphonic/MoppyClassic #114: Moppy: could not find or load main class

Matthew0220, 2015-10-15:
Hi i need help for i don't know what's the problem :(
I get this when i trying to start the project.
[image]

SammyIAm, 2015-10-17:
Huh, interesting.  If I had to take a guess, I'd say you should right click on your project, go to properties, and make sure the main class is set to MoppyUI.java.

A quick Google for that error provides some other potential solutions, but I think it's basically having trouble finding MoppyUI to run.  You could also try right clicking on MoppyUI and choosing "Run this file" directly.

Matthew0220, 2015-10-17:
I tried to single-run only MoppyUI.java and again i get could not get or load main class. There is the photo's...
![nevtelen1](https://cloud.githubusercontent.com/assets/15145392/10558039/d6a9972a-74c1-11e5-8c24-2292295faae1.png
And there's the run settings...
[image]

tg44, 2015-10-17:
Regarding to google:
https://www.youtube.com/watch?v=2vTbonqij_Q
http://stackoverflow.com/questions/18814602/error-could-not-find-or-load-main-class-netbeans-and-java
http://stackoverflow.com/questions/20034377/netbeans-error-could-not-find-or-load-main-class

I saw you are useing hungarian windows. It could be a possible issue if your user name contains áéőúóüö characters and you working on documents or desktop.
If you write me on fb I will try to help you, in our native lang it would be much faster ;)

solidsnake745, 2015-10-17:
To me it doesn't look like it's building the entire project. It states it's only compiling one source file, yet there are more than 20 files in the project.

Try right clicking the project and selecting "Clean and Build". After that try right clicking the project again and hitting "Run". If it's still not working, can you provide a screenshot of the Clean and Build action?

Matthew0220, 2015-10-17:
Tg44 i writed to you on facebook.
Solidsnake i tryed what you typed and this happend.
[image]
"Run"
[image]

Matthew0220, 2015-10-17:
Problem solved!!!
I downloaded java jdk8 and set to jdk8 the netbeans and deleted the vm options from properties-run.
Thx for help!

closed 2015-10-17 by Matthew0220

SammyIAm, 2015-10-18:
Hooray!  Glad you got it solved.  The vm options might be needed to get the serial communication working, but it might not be required (depends on how your system is set up).

Thanks for sharing what you did to get it working though, that'll be helpful for others.