samphonic/MoppyClassic #115: Portable setup idea: plausible?

QwertzHz, 2015-11-16:
I have three Raspberry Pis (B2, B+, B), an Arduino Mega 2560, and eight floppy drives set up and ready to go. First, I want to construct a wooden frame (or similar) to hold the drives and the wiring in a fashion that allows it to be portable (i.e. carryable). Second, I want to use a Raspberry Pi to run the Java program and send the data to the Arduino. Third, I want to connect a MIDI keyboard to the Pi and use it to play notes on the drives. I should note that I plan on powering this as follows: drives, 5V2A adapter; Arduino, Pi USB port; Pi, 5V2A adapter. Both adapters could share an extension cable that could be connected wherever I wanted to put the frame. I'm not concerned about the hardware, though.

Also (and this might be a stretch), I was thinking the drives could divide and conquer key combos, so that a two-key combo would be 4-4 drives and a three-key combo would be 2-2-2 drives, a four-key combo 2-2-2-2 drives, etc.

Anyone know whether this would be doable?

DefectX11, 2015-11-16:
I've been very silent in this project for some time, I haven't had a reason to jump in and discuss, but I do have a great portable setup for the Moppy project! I built mine some years ago but the concept should work fine for anyone. 

I used the following to build it-
8 floppy drives
A 300W standard ATX PSU
An Arduino board
A breadboard
Some IDE cables
Sheet metal from a computer case
An old Sony camcorder case from the 90's (big)

I cut the sheet metal to fit in the bottom half of the camcorder case and mounted the drives, connectors pointing out, to the top of it. I took the PSU, trimmed off the metal to slim it down, cut the unnecessary cables and shorted the ON pin. Power was taken from the molex connectors for the drives. I soldered up the power to each drive and cleaned up the cabling, use the IDE cable to make a really clean wiring system to control the drives and had them all connect to a central breadboard mounted on the bottom. The arduino board wasn't mounted permanently so I could remove the board if necessary for other projects, hence the cable termination at the breadboard. Then all that needed to be done was a cleanup job- cut a hole and mounted the power in port to the outside of the case, below the metal sheet; tested each drive and checked for any excess vibration against the metal; checked the grounding of the PSU to keep it safe and shock free. 

It looks great, folds up to a portable case and has room for expansion in the top half of the case! I had planned to do two things for the top half.
[A] Mount two scanner stepper motor units on top of each other to provide long, drawn out musical notes (floppy step lengths are short, scanners can move for a much longer distance). I found that Canon LIDE scanners worked great for size, but I never got around to checking the sound. 
[B] Mount a Raspberry Pi, LCD screen and make the necessary power and data connections to make this a truly all-in-one unit. This never got past planning because I at this point I had moved to a new home and found I didn't have the time any longer to pursue either plan. 

I still have the whole project sitting in disrepair in the depths of my attic. Maybe I'll finish plan A sometime, by for now perhaps you could use this little bit of wisdom to make up your own version! If you'd like to see more I'd be happy to take some photos, and send you the link to a (albeit crappy and quiet) YouTube video that shows the (not quite) finished project in action. 

As for the keyboard... I can't help you there. The only experience I have with that was using a Korg MicroKey keyboard to live control a floppy drive. Can't recall how I had it set up to do that but it only controlled one drive (the first one in series). Funny, even the pitch shift dial worked!

SammyIAm, 2015-11-18:
@QwertzHz The divide-and-conquer key combos is an interesting idea.  It could definitely be implemented in the DrivePooler.java class as a new strategy type.  The tricky bit is that because the MIDI notes will arrive one at a time (rather than all at once) you'd have to very quickly divide between drives.  The code would look something like (in pseudo-code):

`ArrayList<int> currentNotes;`
`int[] playingNotes = int[8];`

`if (noteOn){`
`currentNotes.add(newNote)`
`computePlayingNotes()`
`} else if (noteOff){`
`currentNotes.remove(newNote)`
`computePlayingNotes()`
`}`

`computePlayingNotes(){`
`<evenly split notes between the 8 playing notes and send these to the drives>`
`}`

You'd have to decide if you were okay with all 8 drives playing the same note if only one key were pressed, and you'd also have to figure out a strategy for when 5+ keys are pressed (do you drop to single drives for each note?  What if it goes over 8?).  Still, probably pretty do-able.

QwertzHz, 2015-11-18:
Yeah, I'm not worried about the actual coding (I spent a year in an
advanced placement Java class) as much as the connection between the Pi and
the Arduino. But I think I'll try to add that strategy sometime soon and
see how it goes. I'll just have to read up on the program.

AmEv7Fam, 2015-11-19:
I think it should be. I'm planning on doing a similar setup myself.

The only potential issue I foresee is finding an ARM serial driver...

solidsnake745, 2015-11-20:
I haven't done it myself, but I'm almost 99% sure this is do-able. Breaking it down into logical components:

1. Running Moppy on Raspberry PI:
  * Moppy is a Java based application. Java can be run on Linux. Raspberry Pi runs Linux. Check.
2. Connecting an Arduino based microcontroller to the PI:
  * Depending on the board, either an FTDI or ARM based USB Serial driver will be needed. Should not be a problem as there are Linux serial drivers out there and a multitude of people using serial communication to Arduino on Linux/Mac.
  * There is one more possibility though it may be difficult to fully implement. The PI has a GPIO header that breaks out some serial communication pins. You could theoretically utilize those for talking to a single microcontroller. The major problems with this is exposing those pins to the Moppy application. They probably won't show up by default in Moppy's dropdowns. You'll have to build something in order to get Moppy to talk through those pins. One more problem is that the PI's GPIO is 3.3v logic so you'd need to level shift it for some Arduino boards as most use 5v logic.
  * Just to add to the previous point, it's definitely do-able. I had an idea to create a sketch for another Arduino to process a MIDI file and send the relevant events over to a Moppy based Arduino in order to make a more portable setup. This was the initial prototype and it worked very well: https://www.youtube.com/watch?v=juwo6D0kjg0. The USB connected to the red micrcontroller board is simply providing power to both the controllers and the single drive. I really need to get back on this, finish it, and commit it to the repo for others to experiment with.
3. Powering everything:
  * Haven't researched or measured this myself, but I've seen 8 drives peak a little over 2 Amps through this person's videos: https://www.youtube.com/watch?v=vZe0_dk4czE. So depending on the number of drives you plan on connecting, you may need a little more juice out of that adapter. An ATX power supply is a fantastic alternative. Not only does it have a high output, but there a number of extra features that you can utilize for your setup:
    * It has 12v, 5v, and 3.3v outputs. You can use that for your drives and powering your PI and any other potential devices you may connect.
    * Those outputs have high current ratings. Much higher than most adapters at a similar cost point. Looking at the cheapest ATX supply on Newegg has the following ratings: +12v @ **16 A**, +5v @ **22 A**, +3.3v @ **18 A**. You'd probably be able to salvage one from a scrap computer too. If that's too bulky, there are smaller form factors available such as LFX or SFX. Just an idea to consider.
    * It allows for digital logic control over the PSU state. You can turn it on and off by grounding one of it's pins. This allows you to turn it on and off programatically.
    * There is standby power that you can tap into to power your PI at all times. So you can keep your PI up and running even if the drives are not being powered.
4. Connecting a MIDI piano to the PI and playing the drives from it's input:
  * Moppy is capable of accepting input from MIDI devices (any MIDI device!) and then interpreting and sending those events to the designated outputs. You just need to ensure the keyboard is recognized by the PI as a generic MIDI device and then it should be selectable within Moppy on the MIDI input option. I've used a keyboard in this way via this specific USB adapter: http://www.amazon.com/VicTsing-Cable-Converter-Keyboard-Adapter/dp/B00ACGMOA6/ref=sr_1_1?ie=UTF8&qid=1448033718&sr=8-1&keywords=usb+MIDI. There are reviews on it indicating people who have been able to use it successfully on Ubuntu so it should work for the PI as well. Theoretically.

As for the pooling strategy you want to implement, I'll defer to Sam's response. He knows that stuff way better than me. I'd be interested to see your resulting code for it. Pass it by me if you don't mind.

Good luck!

QwertzHz, 2015-11-21:
Thanks! I'll look into this and let you guys know if I get anywhere with it
(school is certainly busy).

MuffyTarkin, 2016-09-21:
I've actually been working on getting a portable solution up and running. I actually whipped up something in a free CAD program which has mounting space for 16 floppies. They're arranged in two rows, with four columns in each row and each column being two floppies high. One row is on the top, another row on the bottom. And in the middle is basically a square/hollow rectangle. Within this square/rectangle towards the back are two hard drives, for speakers if you so desire to try and use them.

And then a slide out/pull open panel on the back to access all the wiring.

I have not gotten around to any actual construction of this, but I think it would work very well with the right types of wood.

The first thing I did to function as a step to this was clean up all the wiring. I personally started with hundreds of thin jumper wires for breadboards. That was a nightmare and I never want to go back to that. Setting it all up took about an hour. I "upgraded" the wiring by splicing and soldering things together, and ran with that for awhile but it was still tedious and annoying to diagnose hardware problems. Why isn't that floppy doing anything? Oh, guess the wiring is messed up. Huh, well I can't find any of its wires. Let me spend 20 minutes trying to find them. Oh, they weren't even plugged in.

I noticed solidsnake745 had his own cables that he used. I decided to give that a try, and it's a world of difference. I bought some 34-pin FPC connectors and took some Cat5 ethernet from work, spent an afternoon trimming and stripping, and then wired all the connectors up and terminated the other ends with dupont male pins.

In this process and testing I noticed that all of the grounds for the floppies are common. I never looked up any pinouts or guides for this, I just glanced over an instructables to find the software and got to it. I tested the pins and of course found grounding three of them makes things happen, so I assumed that those pins HAD to have their respective ground going to the Arduino. Nope. Not at all. All of the important pins can go to a single ground on the floppy. So I reduced the total number of conductors from 6 to 3. One ground, one step, one direction. I have the drive select and step wire twisted and soldered at the FPC connector to keep things clean on the other end of the cable.

In fact, I noticed that it seems ground is ultimately not required. I have 16 drives and currently I have not bothered hooking the grounds up to the Arduino and they work fine. I don't know if this is normal or not, but seriously. They're fine. None are out of sync and they all do what they're supposed to.

My next step is to get my own power bus going so I can get away from these stupid molex connectors. I hate these damn things. They always feel like they're going to fall apart and the pliability of the wiring is so stiff that it's very difficult to manage them.

Once I get that figured out then I'm going to work on building that case. Wish me luck friends.

SammyIAm, 2016-09-28:
Well, having some sort of ground absolutely *is* required, but in your case the drives are probably being grounded through the power supply.  In *theory* all the ground pins and the ground cables for power will all be at the same reference voltage, and could safely all be connected together (and in some/many floppy drives they probably are).

Where you run into trouble is if your Arduino tries to pull a pin to "LOW" (e.g. ground), but is grounded at a difference reference voltage so that the drive doesn't see that signal pin going low.  Depending on your power situation and drives this is more or less important.  If it's working for your without the dedicated ground wires though, that's great!

MuffyTarkin, 2016-09-28:
I think you may be right. It's also worth noting that from a hardware perspective, the casing of a floppy is also a ground. It's entirely likely I do not require a dedicated ground bus because I've got all of mine arranged in stacks and touching one another.

I future proofed the wires I made though and have a loose wire on both ends of each cable that I can use for a ground if I ever decide to use it.

closed 2023-08-11 by lpulley