samphonic/MoppyClassic #116: I dont know what to do

Ipolek, 2015-12-09:
what do i need to do?
ant -f C:\\Users\\Olek\\Desktop\\MoppyDesk -Dnb.internal.action.name=run run
init:
Deleting: C:\Users\Olek\Desktop\MoppyDesk\build\built-jar.properties
deps-jar:
Updating property file: C:\Users\Olek\Desktop\MoppyDesk\build\built-jar.properties
Compiling 23 source files to C:\Users\Olek\Desktop\MoppyDesk\build\classes
warning: [options] bootstrap class path not set in conjunction with -source 1.7
C:\Users\Olek\Desktop\MoppyDesk\src\moppydesk\MoppyUI.java:8: error: package gnu.io does not exist
import gnu.io.NoSuchPortException;
C:\Users\Olek\Desktop\MoppyDesk\src\moppydesk\MoppyUI.java:9: error: package gnu.io does not exist
import gnu.io.PortInUseException;
C:\Users\Olek\Desktop\MoppyDesk\src\moppydesk\MoppyUI.java:10: error: package gnu.io does not exist
import gnu.io.UnsupportedCommOperationException;
C:\Users\Olek\Desktop\MoppyDesk\src\moppydesk\outputs\MoppyCOMBridge.java:3: error: package gnu.io does not exist
import gnu.io.NRSerialPort;
C:\Users\Olek\Desktop\MoppyDesk\src\moppydesk\outputs\MoppyCOMBridge.java:4: error: package gnu.io does not exist
import gnu.io.NoSuchPortException;
C:\Users\Olek\Desktop\MoppyDesk\src\moppydesk\outputs\MoppyCOMBridge.java:5: error: package gnu.io does not exist
import gnu.io.PortInUseException;
C:\Users\Olek\Desktop\MoppyDesk\src\moppydesk\outputs\MoppyCOMBridge.java:6: error: package gnu.io does not exist
import gnu.io.UnsupportedCommOperationException;
C:\Users\Olek\Desktop\MoppyDesk\src\moppydesk\outputs\MoppyCOMBridge.java:22: error: cannot find symbol
    NRSerialPort com;
  symbol:   class NRSerialPort
  location: class MoppyCOMBridge
C:\Users\Olek\Desktop\MoppyDesk\src\moppydesk\outputs\MoppyCOMBridge.java:25: error: cannot find symbol
    public MoppyCOMBridge(String portName) throws NoSuchPortException, PortInUseException, UnsupportedCommOperationException, IOException {
  symbol:   class NoSuchPortException
  location: class MoppyCOMBridge
C:\Users\Olek\Desktop\MoppyDesk\src\moppydesk\outputs\MoppyCOMBridge.java:25: error: cannot find symbol
    public MoppyCOMBridge(String portName) throws NoSuchPortException, PortInUseException, UnsupportedCommOperationException, IOException {
  symbol:   class PortInUseException
  location: class MoppyCOMBridge
C:\Users\Olek\Desktop\MoppyDesk\src\moppydesk\outputs\MoppyCOMBridge.java:25: error: cannot find symbol
    public MoppyCOMBridge(String portName) throws NoSuchPortException, PortInUseException, UnsupportedCommOperationException, IOException {
  symbol:   class UnsupportedCommOperationException
  location: class MoppyCOMBridge
C:\Users\Olek\Desktop\MoppyDesk\src\moppydesk\outputs\MoppyPlayerOutput.java:3: error: package gnu.io does not exist
import gnu.io.SerialPort;
C:\Users\Olek\Desktop\MoppyDesk\src\moppydesk\outputs\MoppyPlayerOutput.java:50: error: cannot find symbol
    SerialPort com;
  symbol:   class SerialPort
  location: class MoppyPlayerOutput
C:\Users\Olek\Desktop\MoppyDesk\src\moppydesk\ui\MoppyControlWindow.java:6: error: package gnu.io does not exist
import gnu.io.NoSuchPortException;
C:\Users\Olek\Desktop\MoppyDesk\src\moppydesk\ui\MoppyControlWindow.java:7: error: package gnu.io does not exist
import gnu.io.PortInUseException;
C:\Users\Olek\Desktop\MoppyDesk\src\moppydesk\ui\MoppyControlWindow.java:8: error: package gnu.io does not exist
import gnu.io.UnsupportedCommOperationException;
C:\Users\Olek\Desktop\MoppyDesk\src\moppydesk\ui\MoppyControlWindow.java:262: error: cannot find symbol
    private void initializeReceivers() throws NoSuchPortException, PortInUseException, UnsupportedCommOperationException, IOException, MidiUnavailableException {
  symbol:   class NoSuchPortException
  location: class MoppyControlWindow
C:\Users\Olek\Desktop\MoppyDesk\src\moppydesk\ui\MoppyControlWindow.java:262: error: cannot find symbol
    private void initializeReceivers() throws NoSuchPortException, PortInUseException, UnsupportedCommOperationException, IOException, MidiUnavailableException {
  symbol:   class PortInUseException
  location: class MoppyControlWindow
C:\Users\Olek\Desktop\MoppyDesk\src\moppydesk\ui\MoppyControlWindow.java:262: error: cannot find symbol
    private void initializeReceivers() throws NoSuchPortException, PortInUseException, UnsupportedCommOperationException, IOException, MidiUnavailableException {
  symbol:   class UnsupportedCommOperationException
  location: class MoppyControlWindow
C:\Users\Olek\Desktop\MoppyDesk\src\moppydesk\outputs\MoppyCOMBridge.java:26: error: cannot find symbol
        com = new NRSerialPort(portName, SERIAL_RATE);
  symbol:   class NRSerialPort
  location: class MoppyCOMBridge
C:\Users\Olek\Desktop\MoppyDesk\src\moppydesk\outputs\MoppyCOMBridge.java:108: error: cannot find symbol
        return NRSerialPort.getAvailableSerialPorts().toArray(new String[0]);
  symbol:   variable NRSerialPort
  location: class MoppyCOMBridge
Note: C:\Users\Olek\Desktop\MoppyDesk\src\moppydesk\ui\PlaylistControls.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
21 errors
1 warning
C:\Users\Olek\Desktop\MoppyDesk\nbproject\build-impl.xml:929: The following error occurred while executing this line:
C:\Users\Olek\Desktop\MoppyDesk\nbproject\build-impl.xml:269: Compile failed; see the compiler error output for details.
BUILD FAILED (total time: 1 second)

SammyIAm, 2015-12-10:
It looks like you're missing the NRJavaSerial.jar.  Make sure it's been added to the libraries for the project.

Ipolek, 2015-12-10:
where do i need to put it? its now in C:\Users\Olek\Desktop\Moppy-moppy-advanced\Java\SerialDrivers

Ipolek, 2015-12-10:
no matters fixed it it is working now :)
btw can i use this software on raspberrypi?

SammyIAm, 2015-12-11:
There's no reason why it can't run on a RaspberryPi, though you'd still need to connect your RaspberryPi to an Arduino to control the drives (or take a look at some of the forks from this repo to see if they've made any progress on direct control from the Pi)

closed 2015-12-11 by SammyIAm