samphonic/MoppyClassic #117: I have a problem on MoppyDesk

Breaksreak, 2015-12-30:
When I opened MoppyDesk project, the IDE printed out this error message.

"
Problem: The project uses a class library called "swing-app-framework", but this class library was not found.
Solution: Click Resolve to open the Library Manager and create a new class library called "swing-app-framework".
"

Then, I tried to run MoppyDest, and it printed this error message..

"
Exception in thread "main" java.lang.NoClassDefFoundError: org/jdesktop/application/SingleFrameApplication
	at java.lang.ClassLoader.defineClass1(Native Method)
	at java.lang.ClassLoader.defineClass(ClassLoader.java:800)
	at java.security.SecureClassLoader.defineClass(SecureClassLoader.java:142)
	at java.net.URLClassLoader.defineClass(URLClassLoader.java:449)
	at java.net.URLClassLoader.access$100(URLClassLoader.java:71)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:361)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:355)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.net.URLClassLoader.findClass(URLClassLoader.java:354)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:425)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:308)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:358)
	at sun.launcher.LauncherHelper.checkAndLoadMain(LauncherHelper.java:482)
Caused by: java.lang.ClassNotFoundException: org.jdesktop.application.SingleFrameApplication
	at java.net.URLClassLoader$1.run(URLClassLoader.java:366)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:355)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.net.URLClassLoader.findClass(URLClassLoader.java:354)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:425)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:308)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:358)
	... 13 more
"

How can I solve problem??

SammyIAm, 2015-12-30:
There is a NetBeans plugin called (something like) "Swing Application Framework".  You can install it from the Netbeans Plugins manager.

Breaksreak, 2015-12-30:
Swing Application Framework problem was solved, but the compiler prints out an exception error and the program "runs" indefinitely without a program window.

The exception error
"
java.lang.UnsatisfiedLinkError: no rxtxSerial in java.library.path thrown while loading gnu.io.RXTXCommDriver
Exception in thread "AWT-EventQueue-0" java.lang.UnsatisfiedLinkError: no rxtxSerial in java.library.path
	at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1886)
	at java.lang.Runtime.loadLibrary0(Runtime.java:849)
	at java.lang.System.loadLibrary(System.java:1088)
	at gnu.io.CommPortIdentifier.<clinit>(CommPortIdentifier.java:123)
	at moppydesk.MoppyMainWindow.updateComSelectionMenu(MoppyMainWindow.java:55)
	at moppydesk.MoppyMainWindow.<init>(MoppyMainWindow.java:43)
	at moppydesk.MoppyUI.startup(MoppyUI.java:19)
	at org.jdesktop.application.Application$1.run(Application.java:171)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:312)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:745)
	at java.awt.EventQueue.access$300(EventQueue.java:103)
	at java.awt.EventQueue$3.run(EventQueue.java:706)
	at java.awt.EventQueue$3.run(EventQueue.java:704)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:715)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:242)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:161)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:150)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:146)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:138)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:91)
"

AlexandruCristian, 2016-01-03:
I have probably the same problem, 

 This is when I open the project
[image]
   and
This is when I press run MoppyDesk
[image]
 And is stopping here
(sorry for my bad English, but I'm from Romania and I am 14 years old)

solidsnake745, 2016-01-09:
@AlexandruCristian Need to see the entire error before we can help. Those last few lines don't really tell us what went wrong.

solidsnake745, 2016-03-02:
@AlexandruCristian, do you still need help?

solidsnake745, 2016-05-09:
Closing as there's been no response for over two months. If you're still having this problem, please create a new issue.

closed 2016-05-09 by solidsnake745