samphonic/MoppyClassic #118: Problem with playing Moppy

Breaksreak, 2016-01-08:
When I load a midi file and press start button, the floppy disk only plays some tick-tick noises. I created the Midi file with <a href="https://en.wikipedia.org/wiki/NoteWorthy_Composer">NoteWorthy Composer</a>. What did I wrong??

solidsnake745, 2016-01-09:
There isn't much information here for someone to work here. Can you provide some more detail on your setup so we can try to help you?

In general, your MIDI needs to play notes on the same channel that the floppy drive is on. So if you want a note to play on the floppy on pins 2/3, you need to assign notes to channel 1. Have you tried any other MIDIs that have worked?

AmEv7Fam, 2016-01-09:
It's possible that there's a hardware issue. Double-check your cable connections. It sounds like you might have reversed the Step and Direction pins.

solidsnake745, 2016-03-02:
@Breaksreak, do you still need help with this?

closed 2016-03-11 by solidsnake745