samphonic/MoppyClassic #119: Midi In

Iseeicy, 2016-01-09:
Hey there, I've been having an issue with Moppy desk. I wanted to see if it was possible to use FL Studio's midi out functions to control my floppy setup. When I changed the input mode in the program though, I didn't see anything other than the Real-time sequencer. Heres a picture: https://gyazo.com/2f1d42259d752699f939dd3ffc198d97

Is it possible to do something like this?

solidsnake745, 2016-01-09:
Absolutely. I primarily re-arrange for my videos using FL Studio and have it's output played through the floppies. It's extremely useful.

The problem you're having is that FL Studio doesn't create a MIDI input device to which Moppy can use. You need to loop back the output from FL Studio into Moppy for it to work. If you're using Windows, you can use loopMIDI (http://www.tobias-erichsen.de/software/loopmidi.html) to accomplish this.

[image]

When you load a MIDI into FL Studio initially, all MIDI instruments default to port 0. I setup my loopback device onto port 1 instead. Then I create new instruments, one for each floppy drive/channel, and assign those to port 1. I can then start copying over notes from the original MIDI instruments over into my new floppy channels. This lets me arrange for individual drives, listen to parts of the song I'm working on, solo out drives, etc. In the below image you'll see I what I just described:

[image]

So the original song is in the top 4 Piano instruments and then my floppy drives/stepper motors are in the instruments labeled 1 through 9#2.

Iseeicy, 2016-01-09:
Thank you very much! This has saved me so much time

closed 2016-01-09 by Iseeicy