samphonic/MoppyClassic #120: 16 FDD Setup via Arduino UNO/Arduino Duemilanove

BartMaster1234, 2016-01-21:
I want to do a 16 FDD setup using an Arduino Uno and a Duemilanove. Currently my setup uses the Duemilanove with 8 connected drives connected through COM3. 

Is it possible to use an Arduino MEGA to control all 16, or do I have to use two separate Arduinos? If so, how would I go about setting up 16 drives?

One more thing, is there any support for IDE HDD's or 5.25 FDD's yet? I've seen a couple of videos floating around of set ups consisting of old HDD's and the read/write arm as well as older 5.25 drives.

MoppyDoppy, 2016-01-21:
Hi

yes it is possible to use 16 FDD's. ( MoppyDoppy on youtube, https://www.youtube.com/channel/UCVupT30jd-n1i7CqCu0Zo0g )
I used 2 identical connected and configured arduino's with 8 FDD each. I changed on the java site the MoppyCOMBridge to switch track value (if grater track 8) from 9-16 to 1-8 for the attached arduino.  
If you want to use a mega you need a 5V Version (Mega 2560 as I have). The pin layout for output is a bit different (you can see it in the video). 
If so you need to change the arduino code Moppy.ino to have a greater array. 
Once upon the time I will do it.
Have a look in the arduino code, there you will find your answer for 5.25 drives.

AmEv7Fam, 2016-01-22:
Also, [this](https://github.com/SammyIAm/Moppy/wiki/FAQ#can-i-use-moppy-to-control-a-harddrivescannerprinterc) provides a good explanation why the software can't control HDDs....

themaxx2k, 2016-01-22:
For HDD: you need an audio amplifier and configure MoppyDesk to play MidiNotes lets say channel 1 to AudioMixer. Then you have to disassemble the Harddisk and find connection points to the coil and hook up the Audio Amplifier, which linein goes to your PC Audio Out. Best way  to search for is the ribbon cable connection to the PCB. should be around 4 to 12 Ohm (follow the thicker lines inside ribbon cable). Some HDD have a parkin possition, maybe you need to remove the tiny magnet at the back of the readwrite-arm, cause this may make the arm stuck in "save" position.

solidsnake745, 2016-03-02:
Closing this as you have another issue open for it.

closed 2016-03-02 by solidsnake745