samphonic/MoppyClassic #121: trouble with OSX El Captain :/

makcuber, 2016-01-27:
So my current setup is an Arduino Uno with one Floppy drive attached (just to start out) and I'm trying to run Moppy on OSX El Captain. I already fix an issue with the Arduino IDE itself not working under El Captain where my MacBookPro could no longer see the Arduino over the serial connection (See this: https://forum.arduino.cc/index.php?topic=351999.0).
Currently my issue is that I can't getting Moppy to run. I have the source downloaded and I tried loading the project in IntelliJ (the Java IDE I prefer), but I try to compile it I get this error: 

Error:(26, 54) java: diamond operator is not supported in -source 1.6
  (use -source 7 or higher to enable diamond operator)

I've only used IntelliJ (or Java for that matter) a couple of times before and it seems I am far too much of a Java noob to figure out how to fix this. I have JDK 1.8.72 installed but for some reason the project doesn't seem to compile with the latest Java version. 
I found this google project that I've now imported to Github project: https://github.com/makcuber/moppy-desk-launcher-mac

 I was able to get Moppy up and running, but now I get an error when I try to load a MIDI file:

Java.lang.NullPointerException

Should I just try to install netbeans and build the project from source or am I just missing something with the IntelliJ setup thats resulting in this error?

solidsnake745, 2016-03-02:
Sorry, I missed this completely. I'm not familar with that moppy desk launcher mac project so I can't advise why you're getting that null exception. As for your previous issue, I also can't advise on how to use IntelliJ, but from that error you have to change the version of Java being used on the project. Even you install the 1.8 JDK, it's still backwards compatible and supports previous versions being used.

This seems useful: https://www.jetbrains.com/idea/help/configuring-global-project-and-module-sdks.html

But the error you're getting is just syntax. You can change that line to not use the diamond operator. The diamond operator is just a lazy way of implicitly instantiating a variable.

I can't look up what line is giving you that issue because the error you pasted doesn't mention the file causing the issue. Basically it should look something like this:

`{Collection type}<{Type}> names = new {Collection type}<>();`

It should be changed to also include the type the collection is storing when creating a new instance:

`{Collection type}<{Type}> names = new {Collection type}<{Type}>();`

I realize this could seem pretty vague so let me know the exact line causing you trouble and I can point out what you need to do.

SammyIAm, 2016-04-20:
As @solidsnake745 said, you'll probably want to make sure you're using Java 7 or later (offhand though I don't know how to change that config in IntelliJ).

As for the NullPointer, we can probably help you out there, but we'll need the rest of the stracktrace below the NullPointer so we can see which line has the errant null.