samphonic/MoppyClassic #122: Compiling MoppyDesk

don3194, 2016-01-30:
I do not know how to use NetBeans very well, and I was hoping someone could tell me how to correctly compile this into an executable file of some sort. I have tried some buttons that say "Clean and Build" or "Generate Javadoc" and neither of them don't seem to compile them into one file. Thanks for the Help!
Sincerely,
don3194

SammyIAm, 2016-04-20:
Sorry for the incredible lag: "Clean and Build" should create a MoppyDesk.jar file in the output directory for the project.  Take a look at the output log for the file path.

don3194, 2016-05-22:
Sorry for my incredible lag as well. So I have located the "MoppyDesk.jar" created from clicking the "Clean and Build" button, but when I click to open it, nothing happens. By the way I feel like a complete idiot right now like I am missing something too obvious.

tj2424, 2017-03-29:
try right clicking the file and selecting "run" worked for me

sorry for my incredible lag also

SammyIAm, 2017-06-04:
Yeah, you should be able to just press the "Run" button in NetBeans and it'll run.  If no windows pop up, check the output at the bottom of the screen to see if there was an error.