samphonic/MoppyClassic #123: Power Supply

yigitsen1, 2016-02-17:
Hello!
I want to do this project but I didn't understand one thing. I plan to use 8 drives and I don't have a idea how to connect all of them to floppy. Can you please help?
Thanks,
Yiğit Şen

RedFox1177, 2016-02-17:
Connect them to Moppy or to power? For moppy, you use something like pins
17 and 18 to 2 and 3 on your arduino.

yigitsen1, 2016-02-17:
To power

RedFox1177, 2016-02-17:
> <javascript:_e(%7B%7D,'cvml','notifications@github.com');>>> şunları

You would just plug them into an older PSU. Then, short the green wire on
the big 24-pin connector to a black wire.

yigitsen1, 2016-02-17:
> <javascript:_e(%7B%7D,'cvml','notifications@github.com<mailto:notifications@github.com>');><mailto:
> <javascript:_e(%7B%7D,'cvml','notifications@github.com<mailto:notifications@github.com>');>>> şunları

Correct me if I'm wrong.
İsn't there only one power cable for power supply?
And if there is only one, how can I use it for 8 floppy drives?
One idea was to use breadboard to multiply it, but have no idea at

RedFox1177, 2016-02-17:
I mean like a computer power supply, such as this one:

jazzbassNick, 2016-02-17:
Yep - I used a few "Y" adapters to distribute the regular molex connectors
to the floppies.

yigitsen1, 2016-02-17:
What if we have no Y connectors? And unfortunately in where I live (Istanbul, Turkey) those things are really rare.

iPhone'umdan gönderildi

RedFox1177, 2016-02-17:
You just need a Molex to floppy adapter, or if you have to, solder 12 and 5 volts as described on an online pinout.

yigitsen1, 2016-02-17:
I think that solder can be hard for me. I previously asked that if I simply cut out the connector at the head and then connect the cables from PSU to a breadboard, and then pulling the energy from breadboard, will it be fine?

iPhone'umdan gönderildi

AmEv7Fam, 2016-02-17:
Just to make things a bit easier: I have never seen a 3 1/2" PC floppy drive ever use the 12V lead. (That isn't saying they don't exist, but I do know the 5 1/4" drives usually require both.) Meaning, in theory, you could use a USB brick as a power source for the power supply. Do check your amperage rating on both your floppy drives and your bricks, lest you release the magic smoke out of either.

As far as physically connecting? Yes, you can carefully clip the molex connectors off of old PSUs. I'd recommend _not_ doing it in the middle of the cable, but towards (or, preferrably, at) the opposite end of the cable you're trying to remove.

If you're careful, you might be able to get away with twisting wires and electrical tape, especially with dealing with very-low voltage.

AmEv7Fam, 2016-03-22:
Looking back on this, I don't know about using a breadboard; I personally fear that it might be too much amperage for it. You could use wire nuts, or just electrical tape.

SammyIAm, 2016-04-20:
Bread board should actually probably be okay; there are a fair number of USB-powered floppy drives out there and the power lines in USB cable aren't terribly thick.

Closing this since there hasn't been much activity, but feel free to reopen if there are more questions.

closed 2016-04-20 by SammyIAm

MuffyTarkin, 2016-09-21:
New to github so I don't know if this will reopen or not, but wanted to drop my two cents. Average peak of up to 16 3.25" drives will pull around 2 amps. I actually ran 16 off of a breadboard for quite some time and it worked fine.

So yeah, a USB adapter would definitely work. I would recommend one that's a bit higher than 2 amps though. Then you could just break it put over a breadboard.

Making your own cabling would also be quite easy as you could just get a bunch of 2 pin connectors and terminate your power lines out of the breadboard in them. If you still want the 12v then just do 3.

In fact it'd probably be better to have a literal rail/bus running along all of the floppies that they can plug into. This way you don't have tons of wires running out of the breadboard making it look messy. Some wood and a nice long solid core wire would work great for that.