samphonic/MoppyClassic #124: Dual-Arduino Issues using 16 FDD's / 5.25 Drive Support

BartMaster1234, 2016-02-29:
**Dual Arduino Issues**

I'm trying to set up 16 drives using two Arduinos, but I ran into a couple issues with using my second set of 8 drives, and second Arduino. 

I edited the following piece of code after doing some research from past issue threads to allow the second set of eight drives to function from this thread: https://github.com/SammyIAm/Moppy/issues/35 

Original: `currentPeriod[Serial.read()] = (Serial.read() << 8) | Serial.read();`

New: `currentPeriod[Serial.read()-16] = (Serial.read() << 8) | Serial.read();`

Doing this caused the second set of drives and Arduino to respond, but only play notes through one drive instead of the seven. It sounds like the Arduino is trying to make the FDD (Connected to Arduino Pins 4 and 5) play all the notes being sent to drives 9-16. 

How would I make the second set of drives correctly play all the notes across all the drives instead of just one?

I don't know if this would be causing the issue, but I'm using an Arduino Duemilanove (FDD's 1-8) for my main set, and an Arduino Uno R3 (FDD's 8-16) for the second set. 

**5.25 Support**
 
I'll be honest, I'm not that good with using Arduinos, reading code, and I have the most basic basic knowledge of Java that it's almost laughable. I want to implement 5.25 drives into my setup, but I have no idea what `An array of maximum track positions for each step-control pin.  Even pins
are used for control, so only even numbers need a value here.  3.5" Floppies have
80 tracks, 5.25" have 50.  These should be doubled, because each tick is now
half a position (use 158 and 98).` means software wise, or how to change the code to implement it. 

How would I connect a 5.25 drive hardware wise as well? I have yet to come across a 5.25 drive with a normal IDE connector, instead it has the PCB edge connector. I can tell you'd have to solder directly to the contacts, but I don't know in what matter. 

Does anyone have any experience with setting up 5.25 drives? I honestly have no idea.

solidsnake745, 2016-02-29:
The new code you're using is a bit dangerous in that it's possible to cause an out of bounds reference on the currentPeriod array. Imagine the case that an event for a channel less than 16 came in. That would cause the index used to be negative and crash the Arduino.

To be safe I recommend replacing it with the following:

`byte channel = Serial.read();`
`if(channel > 16) channel -= 16;`
`currentPeriod[channel] = (Serial.read() << 8) | Serial.read();`

This way you won't be able to cause that negative index scenario. That's not what is causing the notes to only play on one drive though.

For that we'll need a little more info. Are you sure your MIDI that you're trying to play has notes playing for channels 9-16? Or maybe you're trying to use drive pooling? Here are two test MIDI files I made for testing my own 16 drive setup. See if these cause all the drives to play.

[16 Drive Test Clips.zip](https://github.com/SammyIAm/Moppy/files/150220/16.Drive.Test.Clips.zip)

The differing model boards should not be a problem. Coincidentally my 16 drive setup also used a Duemilanove and an Uno (though it was R2, not a big enough difference to be a concern).

The `MAX_POSITION` array defines the maximum number of tracks the associated pin's drive has. If you look at the value in the 2 position it says 158. That means the drive connected to pin 2 is to be considered a drive with a max position of 158 tracks. Newer drives (3.5") were higher density and thus have more tracks while older drives were not capable of such density and therefore had larger and less tracks. You'll need to adjust the corresponding pin's max position in order to allow the controller to properly switch directions.

Don't think about this too much. Pick either number and put it in. If you find the drive seems to not switch directions when it should, then you need to tweak that number. If the head reaches the end and takes a sec to switch and start playing again, the number is too high. If the drive is switching directions before it hits the end, the number is too low. 158 and 98 are common values and should apply. But take this chance to experiment and see what happens. You may learn a thing or two in the process and you won't really be harming the drive.

Hard, straight forward example:
- My 5.25 drive is on channel 4
- I've somehow confirmed 98 is the right number of tracks
- Software change should be from this:
  - `byte MAX_POSITION[] = {0,0,158,0,158,0,158,0,158,0,158,0,158,0,158,0,158,0};`
- To this:
  - `byte MAX_POSITION[] = {0,0,158,0,158,0,158,0,98,0,158,0,158,0,158,0,158,0};`

Other than that, connecting to the drive is the same. The step and direction pins should be in similar positions. You don't have to solder directly to the pins. You can buy the connectors and splice in cables instead. I forget the type of connector, but it's purchasable on either Ebay or Amazon. I'll have to look it up later and get back to you on that. But I purchased a bunch of these connectors and used CAT5 cable.

closed 2016-03-07 by solidsnake745

MoppyDoppy, 2016-03-08:
I changed MoppyCOMBridge the sendevent and it works fine
public void sendEvent(byte pin, int periodData) {
        if (pin >= FIRST_PIN && pin <= MAX_PIN) {
            sendEvent(pin, (byte) ((periodData >> 8) & 0xFF), (byte) (periodData & 0xFF));
        } else {
            sendEvent((byte) (pin - 16), (byte) ((periodData >> 8) & 0xFF), (byte) (periodData & 0xFF));
        }
    }