samphonic/MoppyClassic #126: MoppyDesk Will Not Start

doodgerdog, 2016-03-07:
I can't really explain it so here is a screen cap 
[image]

solidsnake745, 2016-03-07:
Looks like it could find or load the main class moppydesk.MoppyUI.

Check and make sure that file is there. Resolve the error that's on that class file (has a red exclamation mark in the file browser on the left).

doodgerdog, 2016-03-09:
So apparently I had an old version of Moppy.
But now that I have downloaded the newest version I am getting 4 errors.
[image]
[image]
[image]
[image]
What should I do to resolve the errors? Sorry I suck at Java.

doodgerdog, 2016-03-10:
Also I haven't changed anything in the file so everything should be there.

solidsnake745, 2016-03-10:
I don't know what the errors are. You should let us know what they say because they could be a wide variety of problems. But taking a wild guess, I'd say it's having trouble importing the GNU library is. That library comes from the included package in the SerialDrivers folder: nrjavaserial-3.9.3.jar. Make sure under libraries you see a reference for it and then make sure that file is in the correct location.

doodgerdog, 2016-03-11:
Alright MoppyControlWindow.java says that it contains files with errors.
And nrjavaserial-3.9.3.jar is in the serialdrivers folder but they still say "Package gnu.io does not exhist"
Should I just re-install java?

solidsnake745, 2016-03-11:
Reinstalling either the JDK or the runtime won't resolve the issue. The library doesn't come from the Java installation. What about the reference I asked about? Is there a reference to that nrjavaserial-3.9.3 in the project (under 'Libraries')?

doodgerdog, 2016-03-11:
nrjavaserial-3.9.3 is in Moppy-moppy-advanced\Java\SerialDrivers. But I don't see a folder in the netbeans project that is titled 'Libraries'.

doodgerdog, 2016-03-11:
is the folder titled "lib"?

solidsnake745, 2016-03-11:
No, it's called Libraries. 

Using your screenshot:
[image]

doodgerdog, 2016-03-16:
[image]
Its not there

solidsnake745, 2016-03-16:
Sorry, have not had time to look at this. Very busy at work. I'll try to look tonight.

doodgerdog, 2016-03-17:
Alright

doodgerdog, 2016-03-20:
Hey solid you still there?

solidsnake745, 2016-03-22:
I somehow managed to get food poisoning the night I said I'd reply. Since then I've been trying to catch up at work since I had to take two days off. Deadlines to be met. Forgot to get back, sorry.

That looks like the problem. Add the nrjavaserial JAR as a library. Right click on libraries and then select 'Add JAR/Folder'. Once you add that you should be able to compile. This is how my Libraries looks.

[image]

doodgerdog, 2016-03-22:
Thanks so much! If I run into any further problems Ill be sure to ask!

doodgerdog, 2016-03-23:
Hey sorry to bother you again. So everything works except for when I press the play button the program says it is playing but the floppy drives aren't moving or responding. Their lights aren't on either. I know its probably a hardware issue but I don't know where else to ask something like this. Thanks.

doodgerdog, 2016-03-23:
I know that both drives are getting power as the power-supply is functioning and I just tested the plugs for the floppy drives with my multimeter.

doodgerdog, 2016-03-23:
I have them wired up in this configuration. http://i.imgur.com/bjPFH15.jpg

RedFox1177, 2016-03-23:
Have you put drive select B to ground?

doodgerdog, 2016-03-24:
No it is wired exactly like it is in the image.

solidsnake745, 2016-04-10:
Sorry, I've been really preoccupied, did you still need help with this?

doodgerdog, 2016-04-11:
Yes I do. Thank you.

solidsnake745, 2016-04-11:
Where is this image from? That wiring is wrong. It shows connections to pins 20 and 22 on the floppy drive connector. The pins you need are 18 and 20. I've updated the image you've linked with the correct information. I don't have a great image editing software, but basically the step and direction pins I've pointed out need to be connected together.

[image]

doodgerdog, 2016-04-12:
I would assume that the ground pins for 18 and 20 go to ground instead of 19 and 20.

doodgerdog, 2016-04-13:
Yay everything is working now except uploading to my UNO. The error report says>
avrdude: ser_open(): can't open device "\\.\COM1": The system cannot find the file specified.

doodgerdog, 2016-04-13:
I just got everything powered up, but there is just one last problem. When I try to upload moppy for arduino it gets half way done and then says;
avrdude: ser_open(): can't open device "\\.\COM1": The system cannot find the file specified.

AmEv7Fam, 2016-04-16:
Sounds like you're in the Arduino IDE to me. If so, double-check your serial port in it; almost sounds like it's trying to grab the computer's built-in serial port (COM1), which most likely isn't what your Arduino is connected to.

doodgerdog, 2016-04-16:
So I uninstalled IDE and installed the newest version and then added the timer one library and this is what its giving me. 

> C:\Users\dodge\Desktop\Moppy\Moppy.ino:8:0: warning: "RESOLUTION" redefined [enabled by default]
> 
>  #define RESOLUTION 40 //Microsecond resolution for notes
> 
>  ^
> 
> In file included from C:\Users\dodge\Desktop\Moppy\Moppy.ino:1:0:
> 
> C:\Program Files (x86)\Arduino\libraries\Timerone/TimerOne.h:42:0: note: this is the location of the previous definition
> 
>  #define RESOLUTION 65536    // Timer1 is 16 bit
> 
>  ^
> 
> 
> Sketch uses 3,658 bytes (11%) of program storage space. Maximum is 32,256 bytes.
> Global variables use 333 bytes (16%) of dynamic memory, leaving 1,715 bytes for local variables. Maximum is 2,048 bytes.
> avrdude: ser_open(): can't open device "\\.\COM1": The system cannot find the file specified.
> 
> 
> Problem uploading to board.  See http://www.arduino.cc/en/Guide/Troubleshooting#upload for suggestions.

AmEv7Fam, 2016-04-17:
What options to you have for Options -> Serial port in the Arduino IDE? Try changing in them.

doodgerdog, 2016-04-17:
There is no options tab that I can find.
And this is also a problem 
C:\Users\dodge\Desktop\Moppy\Moppy.ino:8:0: warning: "RESOLUTION" redefined [enabled by default]

 #define RESOLUTION 40 //Microsecond resolution for notes

 ^

In file included from C:\Users\dodge\Desktop\Moppy\Moppy.ino:1:0:

C:\Users\dodge\Desktop\arduino-nightly-windows\arduino-nightly\libraries\Timerone/TimerOne.h:42:0: note: this is the location of the previous definition

 #define RESOLUTION 65536    // Timer1 is 16 bit

 ^

AmEv7Fam, 2016-04-18:
Possibly, but I'm not sure.

Is this what you're seeing?
[image]

SammyIAm, 2016-04-20:
Just to make sure things are clear:

Arduino Pin | Floppy Pin
---- | ----
2 | 20
3 | 18
GND | 17 & 19*

* Or just one of them, or the ground on the power connector.  In theory all three of those should be connected, so setting any to ground should ground the rest.

The #RESOLUTION issue sounds like it might be a new bug in the Moppy code.  In the Moppy.ino file, try replacing any instance of "RESOLUTION" with "MOPPY_RESOLUTION".  (Looks like just lines 8 and 74)  It seems like in the latest nightly build of Timer (which it looks like you downloaded), that they are also defining a RESOLUTION constant that's conflicting with Moppy's

doodgerdog, 2016-04-21:
Great they are working now! Thanks!

closed 2016-04-21 by doodgerdog