samphonic/MoppyClassic #128: A Question.

doodgerdog, 2016-04-09:
Can this work with the larger older floppy drives?

AmEv7Fam, 2016-04-10:
Yes. I have personally used this software with the 5 1/4" floppy drives, with great success.

They seem to play the lower notes louder (C-1 to B-2) a bit louder than the higher notes (C-3 to B-4), but it varies on the drive.

There seems to be some Arduino tweaks you may need to change, though. (Honestly, I never changed the Arduino code on mine, and I never noticed a difference.)

solidsnake745, 2016-04-10:
Yes, older drives should work. I've also used this software with 5.25 drives and also did not have to change the Arduino code.

The only tweak you should have to do is determining the max position to use. That indicates to the Arduino code when to switch directions. For 3.25 drives, their disks contained 80 tracks which means after traveling that many tracks, you've reached the end and need to switch directions. That's why you see the Arduino code has a bunch of 158s in the max position array. The code pulses the pin twice per cycle in order to achieve one step. Sammy found that cutting off a track lead to smoother direction switches since you didn't travel all the way to the end.

doodgerdog, 2016-04-13:
Okay thats great! But could I run a 5.25 and a 3.25 on the same arduino?

SammyIAm, 2016-04-20:
Yup, you should be able to use the same Arduino.  Like @AmEv7Fam and @solidsnake745 said, you may need to adjust the MAX_POSITION values in the Arduino code for the larger drive though, depending on if it supports high density (80 track) disks or not.

Basically, if you hook it up and you notice that the head on the big drive is going all the way to the end, sticking there for a while, and then eventually heading back home, you'll want to replace 158 with 78 for that drive.

doodgerdog, 2016-04-21:
How do I change the max position on only the larger drive? Sorry I am really bad at coding.
Also It doesn't seem to stick there, it just hits the end of its rails and makes a "click" sort of sound before reversing directions.

SammyIAm, 2016-04-22:
In the Arduino code, there's an array named "MAX_POSITION" with a bunch of 158's in it.  Each 158 corresponds to the step pin for the drive connected to the Arduino.  That is, the 158 that's at index 2 (the third position in the array since the first one is at index 0) corresponds to the drive connected to pin 2.  If the larger drive is, for example, connected to pins 6 and 7, you would change the 158 that's at index 6 in the array.

If it's hitting the end of the rails, it might be going a little too far.  Try 78 to start, but if that doesn't seem to go far enough down the rails before reversing, you can play around with other values.  (You won't break anything by putting in different numbers)

doodgerdog, 2016-04-29:
Thank you Sam!

closed 2016-04-29 by doodgerdog