samphonic/MoppyClassic #129: Arguments

relo1, 2016-04-10:
Hi there, is it possible to specify arguments upon start of Moppy?
I want to start Moppy with MIDI In Port, configured to my Keystation Mini 32 Keyboard and directly connect to Arduino.
...but I have no plan how to realize that :)
relo1

solidsnake745, 2016-04-10:
Currently there is no way of specifying startup arguments to Moppy, but from what you're describing I think we can find you an alternative solution.

Are you imagining the following steps:
1. You start the Moppy solution
2. It displays the application with Input mode set to MIDI in
3. On the MIDI in panel, you're Keystation Mini 32 Keyboard is already selected
4. The application then basically presses connect for you

You can add preferences to be saved and have then have them loaded up at the start of the application. Look at the preferences in the code now and how they work.

relo1, 2016-04-11:
Yes! In the moment Moppy starts in MIDI File-Mode. When I manually change to MIDI IN Port the default MIDI IN device is Real Time Sequencer.
I have to do 3 steps (Input Mode, Device and Connect) to get it up and running.
I'll have a look to the code. Eventually I'll ask for a spoiler ;-) but first I give me a try to find it myself...
Thanks SolidSnake for the fast answer and hint to the solution