samphonic/MoppyClassic #130: Android version

cookins, 2016-04-10:
Hi
Question in next
Is there possible to launch netbeans moppy in Android? I want to play floppy music from my Android device. Thanks.

solidsnake745, 2016-04-10:
Moppy is a Java application. If you can run a Java application in Moppy, then you can theoretically run Moppy out of an Android device. However I'm not sure how well it will translate over. You'll have to implement this yourself.

SammyIAm, 2016-04-20:
You'll need to recompile the application using suitable Andriod serial libraries, and probably need to adapt the UI to something that Android likes.

You may have some luck looking into forks of this project on GitHub that are either ports to Android, or to a command-line version (since that would at least save you the UI trouble).

closed 2016-04-20 by SammyIAm

Lynnden42, 2017-02-05:
hey just asking this before i go out to try and do it myself. Has this been done? and if so can someone send me the file for it? it'd be so much easier for me to just put it on my phone rather than carrying around a laptop on top of my setup.

SammyIAm, 2017-02-05:
I haven't actually heard back from anyone who's done it successfully, but I know a lot of people have expressed an interest.  Check out the forks of this repo to see if anyone made any progress; it's definitely possible.  If you do get it done, let us know here so we can point everyone else to your work :)