samphonic/MoppyClassic #131: Floppys 2 Silent - Any Trick to tune the volume?

CrossCroissant, 2016-04-11:
I got a System with 16 Floppy, some of them are much louder than others. I tried removing the grease which didnt hekp at all... Its barely possible to hear some of the floppys...

doodgerdog, 2016-04-13:
Are the metal covers still on? Taking those off should help

cookins, 2016-04-13:
Or place louder one's on something soft. Sometimes arming diskette eject system doing some volume difference.

CrossCroissant, 2016-04-13:
Our Floppys are all "Naked". We removed the Top Metal Covers. I hope their getting louder when we mount them on a wooden board.

I will also try the eject system.

SammyIAm, 2016-04-20:
I've found they are louder without disks in them.  (Might be a silly suggestion, but a lot of people just stick a disk in and forget about it)  If you're shooting for balance, you could try putting disks into the ones that are too loud to bring them into the range of the diskless ones.