samphonic/MoppyClassic #133: Floppy drives makes loud buzzing sound, no light.

yigitsen1, 2016-05-06:
Hello everyone,

I finally started working on this project and I found out that 3 drives have the same problem: When I connect them to power supply, they all make loud buzzing sound and the activity LED doesn't light up. I would be glad if you could help me.

Thanks.

SammyIAm, 2016-05-06:
Hi there!
What other connections do you have on the floppy drives?  The LED shouldn't
light up when you only connect power anyway, but neither should the drives
make buzzing sounds :P

yigitsen1, 2016-05-06:
I soldered 11 and 12 as drive selection, 17 and 19 as ground, 18 and 20 for step and direction pins.

SammyIAm, 2016-05-06:
If the LED isn't an "activity" LED so much as a "drive-selected" LED.  If you connect power and *only* the drive selection pins, the LED should usually come on.  You might try using 13/14 for the select pins as well to see if those will get the LED to light.

After you've got the LED lit with power, then you can move on to connecting up the step and direction pins, which I think you have backward.  18 is the direction pin, and 20 is the step pin.  This is probably my fault if you followed the video tutorial because it's backward in my graphic.

yigitsen1, 2016-05-11:
I wrote them probably wrong, as I didn't care, but I'Ve solved it. Thanks!

closed 2016-05-11 by yigitsen1