samphonic/MoppyClassic #134: Freezing and "COM sticking"

Lynnden42, 2016-05-10:
As you play a midi and intend to change it from one to another or to change the drive pooling, when you press pause or stop, the drives will freeze, continuing to play the note that was running upon pressing pause/stop. 
As you pause to switch pooling styles the entire program will freeze from time to time, I've had to reset the entire computer as I attempt to relaunch the project I'll find that the COM port is stuck trying to communicate with the arduino even after the board has been disconnecting. the only fix i've been able to come up with is completely restart the computer, relaunch netbeans and the arduino IDE and upload the code again to get things running until the next hiccup.
These issues didn't happen if previous versions of moppy of which i'm wanting to roll back until these issues are fixed.

solidsnake745, 2016-05-10:
Do you have some specific steps to reproduce this? I understand what you're saying in general, but the exact steps would be helpful in reproducing this issue and either fixing or finding the issue. Can you also let us know some other relevant details such as what OS you're using?

For example when you say "as you play a MIDI and intend to change it", does that mean you had MIDI File mode selected, played a song, clicked stop, and then loaded a new sequence?

You shouldn't have to restart your entire computer. When this happens again, have you tried closing Moppy, unplugging and plugging the Arduino back in, and then starting Moppy again?

solidsnake745, 2016-05-10:
I thought about this a little bit more. It seems you're a long time Moppy user. Have you updated the Arduino code lately? We recently made some changes to it that may explain these issues if you haven't.

We now send extra packets to indicate certain events to the Arduino code. If you're using the older code, it won't handle these new events properly that occur on starting/stopping a sequence, connecting, and disconnecting.

Lynnden42, 2016-05-10:
for this i'll be working backward because the later questions are easier to answer. Yes I've tried to unplug the arduino and plug it back in, shut down moppy (but the process stays running according to the Netbeans IDE, i've tired to look through the processes list in task manager looking for something relevent to no avail.
Yes that's exactly what I mean, I'll be playing a midi, want to change it, click stop, and the floppy drives will continue playing until you press the reset button on the arduino. and on occasion the Moppy program will crash.
Currently I'm using a school computer using windows 7 service pack 1 (for some reason).
Concerning your more recent comment, I may, I'll download the MOST recent version of the moppy software and run everything again, this might solve the problem.
Thank you for your time and quick response.

Lynnden42, 2016-05-10:
So yes that was the problem, i was using an old version of the arduino program, i threw a black midi at it using round robin drive pooling for a stress test, it's stuck again, but i think that's just because i there was an overflow.

solidsnake745, 2016-05-10:
Great. If you're not having any other issues, can I close this out?

Lynnden42, 2016-05-10:
ill run some more tests to see if this was indeed the issue.

Lynnden42, 2016-05-10:
nope, still having the same problems, back to square one.

solidsnake745, 2016-05-10:
Alright, if you have a set of steps that consistently reproduce the issue let me know and I can look into it. But as it is, I'm not sure what in particular is causing your issue.

Lynnden42, 2016-05-10:
having a sequence run, pressing stop, then disconnecting. (normally while dealing with a song with a lot of notes (nike necrofantasia or u.n. owen was her)

SammyIAm, 2016-05-13:
How many drives do you have set up, and what pooling settings do you have (a screenshot of your Moppy interface might be easiest).

I'm suspecting drive pooling since you mentioned you were using it at some point, but I'd like to see what settings you have to make sure.

Lynnden42, 2016-05-13:
the COM ports are set to 6 because i use COM ports 5 and 6 for virtual drives and I don't currently have the computer connected to the drives. and currently all i have are 8 drives, the pooling is set to 2-8 because drive 1 is acting up (most likely due to a bad solder job on my part) and it only serves as a temporary fix until i tinker with the solder connections.
[image]

Lynnden42, 2016-06-06:
so upon uninstalling and doing a clean install on a more powerful machine, I threw a black midi "pi" as a stress test at the drives to see what would happen. Upon clicking pause during the song, my computer crashed. with a "WDF_Violation" this new system is running the most recent version of Windows 10 x64 by the way. I think there's some sort of buffer overflow that occurs when you tell it to stop in the middle of a song with a lot of notes in it. I think what's happening is that it's getting a lot of notes thrown at it from the black midi that once you tell it to pause it gets pushed into memory that shouldn't be used by that program so the system crashes. (i'm not a comp sci person but that's all i can think of) https://www.youtube.com/watch?v=Ay8vzCHkgEk this is the midi, use it for testing purposes. just load the file into the java program and let it run into the middle of the song then click pause. this is what caused the crash for me.

MuffyTarkin, 2016-09-21:
I know this is old but I have experienced this as well. I do not do drive pooling, but I have 16 drives and feed it MIDIs which have 16 channels. I've noticed that if I start and stop a song, particularly when there are a LOT of notes, the entire Moppy program will crash and my drives will seize up and they sound like terminators getting ready to kill.

Over the last few months I've gradually added more drives up to the 16 I have now, and around abouts when I got 12 I started encountering this problem. Sometimes I can get it to fix itself by going to shut down so Windows force closes the IDE, and then cancel the shut down and restart Moppy. But sometimes I can't do that, and I just have to restart the entire system to prevent Skynet from rising.

If I let songs play all the way through, I don't run into this. But I'll commonly start and stop songs and play around with the MIDI, and then start and stop it again until I get the right note and channel arrangement.

I've always chalked it up to some kind of overflow or COM error. Sometimes my Arduinos are really touchy and so much as breathing on one of the USB cables will cause it to stop being detected.