samphonic/MoppyClassic #135: bug  i can't run moppy desk

bastechno, 2016-05-10:
hello sorry for my English im french ....
i want to use moppy desk but i can't run it   i have this report from netbeans
ant -f C:\\Users\\bastien\\Documents\\NetBeansProjects\\MoppyDesk -Dnb.internal.action.name=run run
init:
Deleting: C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\build\built-jar.properties
deps-jar:
Updating property file: C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\build\built-jar.properties
Compiling 23 source files to C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\build\classes
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\MoppyUI.java:8: error: package gnu.io does not exist
import gnu.io.NoSuchPortException;
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\MoppyUI.java:9: error: package gnu.io does not exist
import gnu.io.PortInUseException;
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\MoppyUI.java:10: error: package gnu.io does not exist
import gnu.io.UnsupportedCommOperationException;
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\outputs\MoppyCOMBridge.java:3: error: package gnu.io does not exist
import gnu.io.NRSerialPort;
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\outputs\MoppyCOMBridge.java:4: error: package gnu.io does not exist
import gnu.io.NoSuchPortException;
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\outputs\MoppyCOMBridge.java:5: error: package gnu.io does not exist
import gnu.io.PortInUseException;
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\outputs\MoppyCOMBridge.java:6: error: package gnu.io does not exist
import gnu.io.UnsupportedCommOperationException;
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\outputs\MoppyCOMBridge.java:22: error: cannot find symbol
    NRSerialPort com;
  symbol:   class NRSerialPort
  location: class MoppyCOMBridge
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\outputs\MoppyCOMBridge.java:25: error: cannot find symbol
    public MoppyCOMBridge(String portName) throws NoSuchPortException, PortInUseException, UnsupportedCommOperationException, IOException {
  symbol:   class NoSuchPortException
  location: class MoppyCOMBridge
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\outputs\MoppyCOMBridge.java:25: error: cannot find symbol
    public MoppyCOMBridge(String portName) throws NoSuchPortException, PortInUseException, UnsupportedCommOperationException, IOException {
  symbol:   class PortInUseException
  location: class MoppyCOMBridge
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\outputs\MoppyCOMBridge.java:25: error: cannot find symbol
    public MoppyCOMBridge(String portName) throws NoSuchPortException, PortInUseException, UnsupportedCommOperationException, IOException {
  symbol:   class UnsupportedCommOperationException
  location: class MoppyCOMBridge
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\outputs\MoppyPlayerOutput.java:3: error: package gnu.io does not exist
import gnu.io.SerialPort;
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\outputs\MoppyPlayerOutput.java:50: error: cannot find symbol
    SerialPort com;
  symbol:   class SerialPort
  location: class MoppyPlayerOutput
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\ui\MoppyControlWindow.java:6: error: package gnu.io does not exist
import gnu.io.NoSuchPortException;
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\ui\MoppyControlWindow.java:7: error: package gnu.io does not exist
import gnu.io.PortInUseException;
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\ui\MoppyControlWindow.java:8: error: package gnu.io does not exist
import gnu.io.UnsupportedCommOperationException;
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\ui\MoppyControlWindow.java:262: error: cannot find symbol
    private void initializeReceivers() throws NoSuchPortException, PortInUseException, UnsupportedCommOperationException, IOException, MidiUnavailableException {
  symbol:   class NoSuchPortException
  location: class MoppyControlWindow
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\ui\MoppyControlWindow.java:262: error: cannot find symbol
    private void initializeReceivers() throws NoSuchPortException, PortInUseException, UnsupportedCommOperationException, IOException, MidiUnavailableException {
  symbol:   class PortInUseException
  location: class MoppyControlWindow
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\ui\MoppyControlWindow.java:262: error: cannot find symbol
    private void initializeReceivers() throws NoSuchPortException, PortInUseException, UnsupportedCommOperationException, IOException, MidiUnavailableException {
  symbol:   class UnsupportedCommOperationException
  location: class MoppyControlWindow
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\outputs\MoppyCOMBridge.java:26: error: cannot find symbol
        com = new NRSerialPort(portName, SERIAL_RATE);
  symbol:   class NRSerialPort
  location: class MoppyCOMBridge
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\outputs\MoppyCOMBridge.java:108: error: cannot find symbol
        return NRSerialPort.getAvailableSerialPorts().toArray(new String[0]);
  symbol:   variable NRSerialPort
  location: class MoppyCOMBridge
Note: C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\ui\PlaylistControls.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
21 errors
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\nbproject\build-impl.xml:923: The following error occurred while executing this line:
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\nbproject\build-impl.xml:263: Compile failed; see the compiler error output for details.
BUILD FAILED (total time: 2 seconds)

2 years ago i run it but now i don't know why it is not working now 

can you help me please  
[image]

Lynnden42, 2016-05-11:
my guess would either be the jdk version you're currently using vs the version MoppyDesk uses or the new library MoppyDesk uses for serial communications. If it's the latter of the two, just go Java/jre(your version here)/lib/ make sure the RXTXSerial library isn't there, (either completely delete it or move it to a different folder such as your desktop, and make sure the file "nrjavaserial-3.9.3" is there.

bastechno, 2016-05-11:
thank you for your answer  but it isnot working i made what you says 
but netbeans whrite this
ant -f C:\\Users\\bastien\\Documents\\NetBeansProjects\\MoppyDesk -Dnb.internal.action.name=run run
init:
Deleting: C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\build\built-jar.properties
deps-jar:
Updating property file: C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\build\built-jar.properties
Compiling 23 source files to C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\build\classes
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\MoppyUI.java:8: error: package gnu.io does not exist
import gnu.io.NoSuchPortException;
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\MoppyUI.java:9: error: package gnu.io does not exist
import gnu.io.PortInUseException;
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\MoppyUI.java:10: error: package gnu.io does not exist
import gnu.io.UnsupportedCommOperationException;
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\outputs\MoppyCOMBridge.java:3: error: package gnu.io does not exist
import gnu.io.NRSerialPort;
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\outputs\MoppyCOMBridge.java:4: error: package gnu.io does not exist
import gnu.io.NoSuchPortException;
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\outputs\MoppyCOMBridge.java:5: error: package gnu.io does not exist
import gnu.io.PortInUseException;
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\outputs\MoppyCOMBridge.java:6: error: package gnu.io does not exist
import gnu.io.UnsupportedCommOperationException;
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\outputs\MoppyCOMBridge.java:22: error: cannot find symbol
    NRSerialPort com;
  symbol:   class NRSerialPort
  location: class MoppyCOMBridge
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\outputs\MoppyCOMBridge.java:25: error: cannot find symbol
    public MoppyCOMBridge(String portName) throws NoSuchPortException, PortInUseException, UnsupportedCommOperationException, IOException {
  symbol:   class NoSuchPortException
  location: class MoppyCOMBridge
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\outputs\MoppyCOMBridge.java:25: error: cannot find symbol
    public MoppyCOMBridge(String portName) throws NoSuchPortException, PortInUseException, UnsupportedCommOperationException, IOException {
  symbol:   class PortInUseException
  location: class MoppyCOMBridge
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\outputs\MoppyCOMBridge.java:25: error: cannot find symbol
    public MoppyCOMBridge(String portName) throws NoSuchPortException, PortInUseException, UnsupportedCommOperationException, IOException {
  symbol:   class UnsupportedCommOperationException
  location: class MoppyCOMBridge
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\outputs\MoppyPlayerOutput.java:3: error: package gnu.io does not exist
import gnu.io.SerialPort;
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\outputs\MoppyPlayerOutput.java:50: error: cannot find symbol
    SerialPort com;
  symbol:   class SerialPort
  location: class MoppyPlayerOutput
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\ui\MoppyControlWindow.java:6: error: package gnu.io does not exist
import gnu.io.NoSuchPortException;
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\ui\MoppyControlWindow.java:7: error: package gnu.io does not exist
import gnu.io.PortInUseException;
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\ui\MoppyControlWindow.java:8: error: package gnu.io does not exist
import gnu.io.UnsupportedCommOperationException;
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\ui\MoppyControlWindow.java:262: error: cannot find symbol
    private void initializeReceivers() throws NoSuchPortException, PortInUseException, UnsupportedCommOperationException, IOException, MidiUnavailableException {
  symbol:   class NoSuchPortException
  location: class MoppyControlWindow
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\ui\MoppyControlWindow.java:262: error: cannot find symbol
    private void initializeReceivers() throws NoSuchPortException, PortInUseException, UnsupportedCommOperationException, IOException, MidiUnavailableException {
  symbol:   class PortInUseException
  location: class MoppyControlWindow
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\ui\MoppyControlWindow.java:262: error: cannot find symbol
    private void initializeReceivers() throws NoSuchPortException, PortInUseException, UnsupportedCommOperationException, IOException, MidiUnavailableException {
  symbol:   class UnsupportedCommOperationException
  location: class MoppyControlWindow
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\outputs\MoppyCOMBridge.java:26: error: cannot find symbol
        com = new NRSerialPort(portName, SERIAL_RATE);
  symbol:   class NRSerialPort
  location: class MoppyCOMBridge
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\outputs\MoppyCOMBridge.java:108: error: cannot find symbol
        return NRSerialPort.getAvailableSerialPorts().toArray(new String[0]);
  symbol:   variable NRSerialPort
  location: class MoppyCOMBridge
Note: C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\src\moppydesk\ui\PlaylistControls.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
21 errors
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\nbproject\build-impl.xml:923: The following error occurred while executing this line:
C:\Users\bastien\Documents\NetBeansProjects\MoppyDesk\nbproject\build-impl.xml:263: Compile failed; see the compiler error output for details.
BUILD FAILED (total time: 0 seconds)
[image]
[image]
[image]

bastechno, 2016-05-11:
[image]
[image]
two foto ...

SammyIAm, 2016-05-13:
NetBeans may not be using that particular JRE to launch Moppy.  Check your NetBeans settings to see which JDK is being used and try putting the nrjavaserial there instead.

If the last time you ran it was two years ago, have you downloaded the most recent version of Moppy?  It's possible some changes have been made that will help you out.