samphonic/MoppyClassic #137: RXTX Driver Error in NetBeans

KoNexD, 2016-05-29:
Hello guys and girl :laughing: 
NetBeans is throwing some errors. Maybe someone can help to fix my problem.

First of all:
-The Hardware is wired correctly (Floppy light + motor is working fine)
-Arduino is working fine (LED is on)
-Arduino Software is working and upload is successful (orange LED blinking etc.)

NetBeans is making some problems. I installed and deinstalled the RXTX Drivers multiple times (old and new x64 versions) but they won't work for me. I deinstalled my old java 7 jdk and installed the newest version of the JDK + Netbeans Bundle. RXTX Drivers and dll's are in the correct places. But if I want to run MoppyDesk in NetBeans I get these errors:

"ant -f D:\\Documents\\NetBeansProjects\\Moppy-moppy-advanced\\Java\\MoppyDesk -Dnb.internal.action.name=run run
init:
Deleting: D:\Documents\NetBeansProjects\Moppy-moppy-advanced\Java\MoppyDesk\build\built-jar.properties
deps-jar:
Updating property file: D:\Documents\NetBeansProjects\Moppy-moppy-advanced\Java\MoppyDesk\build\built-jar.properties
compile:
run:
**Exception in thread "AWT-EventQueue-0" java.lang.NoSuchMethodError: gnu.io.RXTXCommDriver.getPortIdentifiers()Ljava/util/Set;
	at gnu.io.NRSerialPort.getAvailableSerialPorts(NRSerialPort.java:116)
	at moppydesk.outputs.MoppyCOMBridge.getAvailableCOMPorts(MoppyCOMBridge.java:108)
	at moppydesk.ui.ChannelOutControl.initComponents(ChannelOutControl.java:133)
	at moppydesk.ui.ChannelOutControl.<init>(ChannelOutControl.java:25)
	at moppydesk.ui.MoppyControlWindow.setupOutputControls(MoppyControlWindow.java:80)
	at moppydesk.ui.MoppyControlWindow.<init>(MoppyControlWindow.java:60)
	at moppydesk.MoppyUI.startup(MoppyUI.java:50)
	at moppydesk.MoppyUI$2.run(MoppyUI.java:98)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatch
Thread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)"**

Used Programs:
-NetBeans IDE 8.1
-Java Developement Kit (jdk1.8.0_91)
-Arduino Software 1.6.9

Used Files:
"Moppy-moppy-advanced" by SammyIAm (https://github.com/SammyIAm/Moppy)
RXTX Drivers by Trent (old and new Versions but no one worked for me)
Timerone is in the right place too 

Some Screenshots:
[image]
[image]
[image]
[image]

I hope someone encountered this error before and is able to help me out.

Thanks, KoNexD

InsaneByte, 2016-06-23:
I had the same problem. I fixed it reinstaling netbeans and the JKD, I don't know very well where is the problem but I think is something with the configuration of netbeans. (sorry 4 my horrible english xD)

KoNexD, 2016-07-01:
So I installed win 10 on my laptop with a new SSD and the latest Java and NetBeans Version. It's working fine now :D I don't know why my desktop pc dislikes the rxtx drivers so much. Also sry for incorrect grammar/ bad english ;D

KoNexD~