samphonic/MoppyClassic #139: Won't run....

gwinters23, 2016-07-04:
This is what I get. I don't know much of anything about netbeans. :(

ant -f C:\\Users\\Gary\\Desktop\\MoppyDesk -Dnb.internal.action.name=rebuild clean jar
init:
deps-clean:
Updating property file: C:\Users\Gary\Desktop\MoppyDesk\build\built-clean.properties
Deleting directory C:\Users\Gary\Desktop\MoppyDesk\build
clean:
init:
deps-jar:
Created dir: C:\Users\Gary\Desktop\MoppyDesk\build
Updating property file: C:\Users\Gary\Desktop\MoppyDesk\build\built-jar.properties
Created dir: C:\Users\Gary\Desktop\MoppyDesk\build\classes
Created dir: C:\Users\Gary\Desktop\MoppyDesk\build\empty
Compiling 6 source files to C:\Users\Gary\Desktop\MoppyDesk\build\classes
warning: [options] bootstrap class path not set in conjunction with -source 1.5
C:\Users\Gary\Desktop\MoppyDesk\src\moppydesk\MoppyUI.java:7: error: package org.jdesktop.application does not exist
import org.jdesktop.application.SingleFrameApplication;
C:\Users\Gary\Desktop\MoppyDesk\src\moppydesk\MoppyUI.java:13: error: cannot find symbol
public class MoppyUI extends SingleFrameApplication{
  symbol: class SingleFrameApplication
C:\Users\Gary\Desktop\MoppyDesk\src\moppydesk\MoppyUI.java:20: error: cannot find symbol
        show(mainWindow);
  symbol:   method show(MoppyMainWindow)
  location: class MoppyUI
C:\Users\Gary\Desktop\MoppyDesk\src\moppydesk\MoppyUI.java:17: error: method does not override or implement a method from a supertype
    @Override
C:\Users\Gary\Desktop\MoppyDesk\src\moppydesk\MoppyUI.java:27: error: cannot find symbol
        launch(MoppyUI.class, args);
  symbol:   method launch(Class<MoppyUI>,String[])
  location: class MoppyUI
Note: C:\Users\Gary\Desktop\MoppyDesk\src\moppydesk\MoppyMainWindow.java uses unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
5 errors
1 warning
C:\Users\Gary\Desktop\MoppyDesk\nbproject\build-impl.xml:923: The following error occurred while executing this line:
C:\Users\Gary\Desktop\MoppyDesk\nbproject\build-impl.xml:303: Compile failed; see the compiler error output for details.
BUILD FAILED (total time: 0 seconds)

WhiskerWaffles, 2017-03-19:
i've had the same problem, so what i figured out was that i was missing NR.JavaSerial.jar

SammyIAm, 2017-06-04:
You need to install the "Swing Application Framework" for NetBeans.

closed 2017-06-04 by SammyIAm