samphonic/MoppyClassic #141: Floppy Drive Amplification

Lynnden42, 2016-08-26:
Alright so i'd like to keep everything as low-tech as possible on this one. This doesn't concern the software of the moppy controller or anything like that, It's just that the drives themselves are awfully quiet. I have them on a 3/4th's inch plank of wood so it's transportable. I'd like to not get microphones or anything else involved if at all possible. I have an old clamshell computer case that I've gutted would I get something louder if I transferred the drives to this case? (which has an inner metal case and outer plastic case)

MuffyTarkin, 2016-09-21:
If you want to amplify the sound, there's really no way to do it other than changing the acoustics of how you have your drives arranged. There are multiple ways to do this. I got all of my floppy drives from my job. I work on computers and I pulled them all out of junkers that we were sending off to our e-cycler. They're all different brands, different dates of manufacture, etc. And they all sound different.

Some are louder than others. Some can play higher notes better than others. Some only really play low notes even though the notes they're getting are in the higher range. Others are just stupid quiet, and can barely be heard. 

It seems that if you keep the external casing on instead of removing it, that can potentially improve sound quality. Especially if you decide to stack them up. The vibrations of the motor will more easily pass through the casing and any floppies ontop of it that have their casing as well will also have those vibrations passed through. You'll basically achieve a mass resonance effect and greatly improve sound quality.

Apart from that, there's the obvious. If there's a particular element in a song you want to have louder than others, just copy that part to other floppies with a MIDI editor. That's the quickest way to do it, but it's important to think about which ones will be playing what relative to the channels you're pasting over. If you've got them stacked and they have their cases on, you'll want basses and the like closer to the bottom of each stack. That will make them resonate very well. Midrange notes and high notes work well closer towards the tops of each stack.

If you do not stack them, and instead have them spread out, you should look into trying different wood. Plywood is fine but different types of wood will have different acoustic attributes with how they handle vibrations. It's important to remember exactly how the floppies are producing those cool sounds; it's vibrations. Look into woods that are commonly used for creating instruments like violins or guitars, or heck, even subwoofer boxes and try different thicknesses of those.

SammyIAm, 2016-09-28:
Make sure you don't have actual floppy disks in the drives (as this dampens the sound); or alternatively put floppy disks in any drives that are too loud to help balance things out.

There's a YouTube video somewhere (no link, sorry) of someone who glued one of those red Solo cups to each of the read heads on their drives.  The result was surprisingly effective in terms of amplification, and it looked pretty fun too.

Otherwise, as @MuffyTarkin said, try experimenting with some different things to see if you have success, but they are generally just really quiet (which is a feature of the drives if you're not trying to make music with them)

Renarix, 2017-12-02:
https://www.youtube.com/watch?v=X4pi8UT3XAI