samphonic/MoppyClassic #144: "Floating" pin states?

MuffyTarkin, 2016-10-01:
This isn't necessarily completely Moppy related, but it's something I've encountered when using it. I have my stuff set up so the LED activity lights only light up when receiving a step signal, so that they're only lit when each floppy motor is moving. I have used this same method to make the fluorescent bulb in an old HP scanner work on the same principle by sending the motor step signal to a transistor on the bulb inverter.

However, something strange I've noticed (not just with the bulb, but also with the LEDs on the floppies) is sometimes the lights get "stuck" and remain on despite no motor instructions being received. I'm assuming this is caused by some kind of a floating state with the Arduino's microcontroller. When it switches pin states, I'm sure there's a point where the pin in specific becomes "floating", that is, it's not disconnected, but is not outputting anything.

Is there any way to test this and if so, fix it? I don't care too much with the floppy LEDs but with the scanner and how bright the light is it can be somewhat annoying.
[https://www.youtube.com/watch?v=qG4DH-XoM8o](url)

In this you can clearly see what I'm talking about. It's not very often when it occurs but if you watch the scanner bulb you can see it remain powered even though the motor itself is not moving. The way it's wired is it runs to a transistor which flips open to allow power through the rest of the inverter every time it receives an output voltage from the arduino.

I know it's not related to just the scanner and the wiring because I've even seen it occur with the floppy LEDs. Just curious if there's a solution to this. I'd imagine I'd have to throw in a pull up/pull down resistor in line with the output signal to keep the pin from entering a "floating" state, but I'm unsure what kind of resistor would be ideal or where to even place it (between the arduino and my breadboard? between the arduino signal and the transistor?)

solidsnake745, 2016-10-02:
Heads up, I'm typing this answer up my phone so it may be overly brief. 

What you're running into is not exactly a floating state, but the pin is being left in a high (on) state. The reason why is because the code works by toggling the pin state at a given rate until it receives a command to stop. That stop command could come in at anytime and it can happen before the pin gets set to low (off).

Floppy drives, and stepper motor controllers in general, work by stepping the motor when it sees a transition from low to high. There isn't a formatted message going to them that causes them to operate. The Moppy Arduino code relies on this and thus toggles the signal at a given frequency in order to step the motor at that rate to play a note. 

I ran into this issue myself when I wired in LEDs for my own floppy drives. What you would need to do is write code that makes sure to set the given pin to low at the same time it stops playing a note.

I mentioned it before, but I'm writing an Arduino library for this and that's one of the features I've included. I'm not quite ready to release it publicly and I don't have a sketch that works with Moppy just yet. However, if you wanted I could give you the code in the state it's in now and you could see how I've implemented it if you want. Once I get back home today I'll try to remember to link it here.

solidsnake745, 2016-10-02:
[MIDIStepper.zip](https://github.com/SammyIAm/Moppy/files/505050/MIDIStepper.zip)

This is what I have so far. In particular, look at MIDIStepper.cpp - resetChannel. That function gets called whenever it sees the current note is 0. To the code, 0 means the device has stopped playing a note, but is awaiting a reset before it can be assigned another note. resetChannel then gets called to set the step pin low and change a few other properties.