samphonic/MoppyClassic #146: MP3/WAV playback from Moppy?

MuffyTarkin, 2016-11-29:
So the step output pins carry tones with them. I've used this to make hard drives play notes. I was wondering if it would be at all possible to change this to an MP3 or WAV track and have that play as its own channel in sync with the actual MIDI sequence.

I'm looking to experiment with the hard drives and I have a few MOGG files (basically multi channel MP3 with every voice on its own channel) and thought it would be neat to have actual song vocals play back through the hard drives instead of the waves they do now.

I can achieve this with audio mixing but I would love to also have it be a real time thing. Any ideas on where I can start looking?

SammyIAm, 2016-12-01:
As Moppy is currently written, the control of the MIDI playback (and therefore where you would do any sort of MIDI/audio syncing) is in the PC Java application.  You could in theory try to pass Audio to the Arduino and have it send that out to the hard drives, but I think you're much better off at that point just hooking the audio output of your PC to the hard drives (through an amplifier if you're not getting enough oomph from your PC's built-in output).  You can hook one HDD to each channel, so if you've got a motherboard or soundcard that supports surround sound, you have a lot of potential channels to work with.

If you're stuck on having the Arduino drive the HDDs, you could grab an Audio playback library for Arduino and add an additional command to the protocol that allows the Java app to start playback in sync with the MIDI file.  However I would have concerns that trying to play audio *and* generate the signals to drive the floppies might be a bit much.

MuffyTarkin, 2016-12-02:
I can definitely just hook them right into my PC, was just wondering if I could change the program around a bit to also allow audio syncing along with the actual MIDI, which looks like it can be done. I don't know a lot about Java but it would be a neat project to try.

Thanks!

closed 2016-12-02 by MuffyTarkin