samphonic/MoppyClassic #147: Using MOPPY with other Stepper Motors

rk9109, 2016-12-28:
To be brief, I was wondering how Moppy could be extended to be used directly with stepper motors. I am currently using 4 NEMA-17 Stepper Motors with an Arduino and 2 Adafruit Motor Shields. Has anyone else experimented with stepper motors (I know solidsnake745 has used them in his recent videos)  and would have any advice on how to proceed?

I apologize for the vague post, but would appreciate any help.

solidsnake745, 2016-12-28:
Hey, no worries about asking a vague question. We can figure it out together.
I took a look at the documentation for using stepper motors with that shield: https://learn.adafruit.com/adafruit-motor-shield/using-stepper-motors

You will need to modify the Moppy Arduino code in order to make it work with their library. Their library lets you specify an amount of steps to take and in what direction which is fundamentally different from how Moppy works. Moppy works by toggling a step signal pin at a given frequency. 

So you could theoretically replace the code that toggles the pin with code from the library that moves the motor one step in some hard coded direction. One caveat though is that you would need to change the code to only step half the time since the original code is going to execute the toggle pin function twice per step. This is because Moppy only toggles the current state of the signal and stepper motors only step on either the falling or rising edge of that signal. I forget which one it is. It's not exactly straight forward, but it's doable.

Making all those changes to the code though seems a bit much. Personally I instead decided to go the hardware route. I picked up a bunch of these [stepper driver chips](https://www.amazon.com/Elegoo-Stepstick-Stepper-Printer-Robotics/dp/B01GJJGRF2/ref=sr_1_1?ie=UTF8&qid=1482911265&sr=8-1&keywords=a4988) from Amazon that basically gives you a step and direction pin to interface with. At that point, your stepper motor now looks exactly like a floppy drive in terms of it's interface and you can use Moppy as is; no code changes necessary. Simply hook up the step pin from the driver chip to the step pin on the microcontroller (pins 2, 4, 6, 8, etc.). Here is a really good resource I used when figuring out how to wire up the driver chip: https://www.pololu.com/product/1182

Take a look at those two options and let me know if you need more help.

Lothean, 2017-08-14:
Hey, 
Thanks for the help, everything works with the given instructions, but how can we change the amount of steps for our motor in the moppy code ? The numbers are unclear, they talk about tracks
Thanks