samphonic/MoppyClassic #149: My drives are 100% silent.

Boblit67, 2017-01-18:
When I run the program the drives all start moving, but all they do is quietly tick at the same note. Is it an issue with my drives? The heads move but there is absolutely no buzz from the motors.

SammyIAm, 2017-01-19:
If they are just ticking slowly, you likely have the STEP and DIRECTION
pins mixed up.  Try switching your even/odd pins (e.g. 2 <-> 3) and see if
that makes a difference.

Boblit67, 2017-01-19:
Ill try that. I double checked the pinouts and such but maybe I got it wrong when I plugged in the whole mess.

Boblit67, 2017-01-19:
They seem to be ticking at the right tempo though, but only ticking

Boblit67, 2017-01-19:
Also the Arduino code has been a pain to upload. Half the time it uploads fine, half the time it says its missing a file and cant contact the Com.

Boblit67, 2017-01-19:
I figured it out. I have DC motors on my drives :\. Any idea where I can order 5 of the right kind?

closed 2017-01-19 by Boblit67

SammyIAm, 2017-01-19:
I see you closed this but opened another issue-- I'm not sure I understand what you mean about DC motors... the drive motors definitely should be DC (they don't make AC floppy drives (thank goodness)).

If you're concerned the drives aren't correct, could you post a picture of them here?