samphonic/MoppyClassic #150: Still not making any noise. What am I doing wrong?

Boblit67, 2017-01-19:
So I swapped the wires like suggested and it didn't run at all. Only way I can even get it to run is by grounding the light to. When the drive is wired how the instructable shows it will not work at all. The arduino also keeps uploading the program inconsistently. Sometimes it works, sometimes it says files are missing and it can't reach the device. Is it a hardware issue or is it a software issue?

Boblit67, 2017-01-19:
When the light is grounded to itself. ***

SammyIAm, 2017-01-19:
First of all, you only need to upload the Arduino program once unless you're making changes to the code (which there's no reason to do yet if you can't get the drives working).  Once the original code has uploaded successfully once, you should just leave it.

I'm not familiar with the instructable page, so I'm not sure what it's asking.  Could you provide a diagram showing how you have things wired currently?

Boblit67, 2017-01-19:
Yeah. Let me find it. Ill give a description until I can update it with a diagram. I have pins 11 and 12, the light/drive select pins, hooked together so it's grounded to itself. I have the two ground wires, 17 and 19 hooked into the arduino ground port. Then 18 is hooked to port 3 and 20 is hooked to port 2 on the arduino, maybe thd other way around but I know those are hooked up to the right pins, I triple checked to make sure even though the head only moves when they're set up one way. If I hook pins 12 and 18 together into the arduino and ground 11 with 17 and 19 into the arduino it won't run. I've tried about 8 or 9 different FDDs and every single one does the same thing.

Boblit67, 2017-01-19:
https://www.google.com/amp/www.instructables.com/id/Musical-Floppy-Drives/%3Famp_page%3Dtrue?client=ms-android-hms-tmobile-us

Here is the guide Ive been using. The wiring section should have pictures and diagrams.

SammyIAm, 2017-01-19:
Okay, so, let's narrow down potential problem components to figure this out.

First let's focus on the drive itself.  Connect pin 11 to pin 12.  Connect pin 18 to ground (pin 17 is convenient for this).  Connect a wire to pin 20, and tap it repeatedly to ground.  Each tap should cause the head on the drive to move one step toward the center of the platter.  If you disconnect pin 18 from ground, each tap should cause the head to move toward the outside of the drive.  If this isn't working:
* Try connecting pin 13 to pin 14 (Drive select A) instead of 11 to 12 (Drive select B).
* Make sure you have the even and odd sides of the plug straight (maybe your drive is wired upside down).
* Failing that, something's up with your floppy drive; try a different model drive.

Next let's check to make sure the software side of things is working.  When you connect the Arduino to the PC and run Moppy, you should see the RX light on the Arduino blink each time a note in the song you're playing starts or stops.  If you have access to a very simple song (thing Mary Had a Little Lamb), you can confirm this is the case.  If not:
* Make sure you have the correct COM ports and channels selected in Moppy

And finally, the wiring between the Arduino and the floppy drive.  If tapping pin 20 to ground was successful, connect pin 20 to pin 2 of the Arduino, and pin 18 to pin 3.  Connect 17 and 19 to the ground as you've done before.  If you move the floppy drive head into the center of its travel and then reset the Arduino, the Arduino should move the head back to its starting location when it resets.  If it doesn't do this:
* Your sketch may not have uploaded correctly.
* Something may still be off with the wiring.
* If you don't mind posting a picture of your wiring, perhaps someone can spot what's off.
If it *does* reset the head, then your wiring is correct.  If you're still not getting anything when you play a song, it's something to do with the particular MIDI file you're using or your configuration in the software.  Post a screenshot of the Moppy interface and we'll see if we can spot anything out of place.

Boblit67, 2017-01-19:
They do reset, the light comes on and flashes with each note, the heads all move fine. Just absolutely zero sound other than a tick and every note causes the motor to move one step only. Maybe my arduino program didn't upload correctly?

Boblit67, 2017-01-19:
I wont have access to the setup again until Monday so we'll see where it goes from there. From my unprofessional observations, it seems like the Arduino isnt correctly designating what speed or frequency the motor should move at. It moves one single tick every single note, regardless of the MIDI. When I had all 5 drives set up it even delegated notes properly and the heads moved, just no sound.

SammyIAm, 2017-01-19:
Huh, that's bizarre.  The code that causes the drives to reset to their starting points when the Arduino is reset is the same code that would make them play notes during a song (they reset to their original positions at 200Hz (a slightly-sharp G)), so it doesn't seem like there's any issue in communication between the Arduino and drives.

Are you using the sample MIDIs that came with the project?  A screenshot of your Moppy UI might still be helpful when you get back to the setup.

Boblit67, 2017-01-19:
Ive tried sample midis, my own, everything. The motors are 100% quiet, only the tick of the head moving, even when resetting. Ive tried everything

SammyIAm, 2017-01-19:
> even when resetting

... Whatever sound you hear when they are resetting is the most sound you're going to get from the drives; that *is* the sound they make.

I think I'm misunderstanding something here, because if they're resetting when you start up, they *will* be making noise.  Maybe a video would make it easier to understand what you're experiencing?

Boblit67, 2017-01-20:
Ill get a video on Monday. They reset silently. The only noise is the tick of the head. They never produce anything remotely close to a buzz or note. They make the same barely audible tick that I dont even think is from the motor. Its always the same exact tick from every drive, every time it moves. Maybe it's an issue with the drives? Are there different types of FDDs with silent motors?

Boblit67, 2017-01-20:
The heads also always move at the same exact speed, no variation at all.

Boblit67, 2017-01-23:
https://vid.me/SVQs
https://vid.me/aIuK

Here are videos of it resetting and running for about 30 seconds

SammyIAm, 2017-01-23:
Alright, so, good news: the reset looks normal, and it's not silent (quiet yes, but you can still the *bvvrrrrt* of the motor as it moves back).  This means the Arduino software is correctly uploaded, and the wiring between the Arduino and floppy drive is all correct.

Bad news: No idea why it's not working normally when you're playing a song.  As I said above, the correct reset means your hardware and firmware is all fine, so it's probably something with either the Java application or the serial communications.  If you send a screenshot of your Moppy UI I can see if anything stands out, but otherwise I'm a bit stumped.  Try a different computer if you have another available?

Boblit67, 2017-01-23:
Ill try. I don't hear any buzz, but huh. Do I just happen to have 6 really really quiet drives?

SammyIAm, 2017-01-23:
Nah, most floppy drives are actually pretty quiet (which is a "feature" if
you're actually using it to read disks), most videos you see are using
microphones really close to the drives to amplify the sound.  Try this:
close your eyes and listen to the reset video you posted.  Can you hear
when the drive resets?  You should be able to hear when it starts moving
and when it's finished, and the sound in the middle is where the note will
be.

<https://github.com/SammyIAm/Moppy/issues/150#issuecomment-274630998>, or mute
the thread