samphonic/MoppyClassic #151: compiled version?

Lynnden42, 2017-02-05:
is there any way that this can be converted into... idk a compiled file or something akin to a .exe? I'm sure there's some reason this isn't done but I'm wondering if it can be done as it'd make the setup really easy (in comparison to installing an IDE and compilers etc etc to get it all running)

Also, I know this is a closed issue here but I'm gonna take a stab at porting this to android. any help with what i'll need to do to get that ball rolling would be great.

DefectX11, 2017-02-05:
I bid you good luck. Android devices have different kernels to handle the different hardware, which means you'll have to find Arduino drivers and compile them for your specific kernel. A quick Google search shows that you can run an Uno from an Android device, but this is half of the equation. Porting the software should be hard but you'll have to overhaul the UI to make it convenient on small devices, unless it's tablet only.

As for a compiled version- you'd have to write it in C# for a compiled version in .exe format. I can't provide much insight on that because I don't program for Windows.
But, as I see it, this project is aimed for people who already have an IDE installed, or would want one anyway. If this is the only situation when you'll use an IDE, then...

Lynnden42, 2017-02-05:
I just used .exe as an example file extension. really, and yeah, i installed netbeans (a few times on a few different computers) for the sole intent on using them for this project, and each time it's taken me multiple hours to get setup. I know that's just part and parcel of setting up this sort of thing (and part having really crappy computers to work with) but it's just frustrating.

concerning the port, i'm working with a Motorola Maxx 2, (I assume you mean the specific model of android i'm working with) and if that's the case, that's gonna suck. as for the UI, couldn't I just divide the system into two parts? (the drives and com ports on one screen, and the music control on a seperate? 
could I just work in some way to idk zoom in on the current ui to make it easier? to work on some of the smaller stuff?

Lynnden42, 2017-02-05:
Also, fair warning about the porting thing, I'm not a skilled programmer, and I have... very little experience coding in java. I've really only worked in c++, which, yes, I know is an ancestor to java but I know it's rather like apples and oranges.

DefectX11, 2017-02-05:
Since you'll be doing the port, you can make the UI look however you want. The UI won't port over directly, so you'll have to remake it anyway. Just make sure the screen elements hook up to the logic and you can make it as complex or as simple as you'd like.

Lynnden42, 2017-02-05:
It occurs to me that I might be "slightly" over my head on this.

DefectX11, 2017-02-05:
Put it this way- there's a reason it hasn't been done yet.

Lynnden42, 2017-02-05:
eh... i've been looking for a reason to develop a dependency on alcohol for a while now. 
so suppose I create my ui then drop the moppy project files in and just wire everything up to get the ui to talk to the moppy software?

SammyIAm, 2017-02-05:
Now hang on there, before you go getting too worried about this being really hard, it's not actually too far from doable.  There are three key components / things you'd need to accomplish, but first I'll address the .exe bit:  Since it's written in java, there's not a traditional "executable" that's compiled so much as byte code that's executed by the Java Virtual Machine (JVM); I'll not go into too much detail here, but the short story is you don't actually need NetBeans (you can just compile the code on the command line using `javac`), and since Android more or less runs Java, you don't really need to "port" anything at all.

The three things you'll have to do are:
* As you mentioned, you'll need to divide the current Moppy program into the backend functional piece and the UI.  This is mostly already done; a lot of the actual communication with the Arduino is done in separate classes, so this shouldn't be too hard.  If you want a UI for Android you'll need to build that yourself separately, and while I have no experience in this area myself, it shouldn't be any harder to do for Moppy than for any other Android app
* Figure out a way to actually connect an Android device to the Arduino.  I'm sure Google can help with this somehow-- surely someone's done it.  All Moppy needs is serial communication, which should be a pretty basic ask.
* Finally, you'll need to replace RXTX libraries in Moppy (that handle the serial communication) with libraries that support Android.  Again, this shouldn't be too bad as I assume that someone has already solved the Android->Serial communication problem.

I unfortunately don't have the time, resources, or motivation to do this myself, but I'm more than happy to answer any specific questions you have along the way, or try to point you in the right direction.

For a start, check out some of the forks of this project on GitHub (the "forks" link on the top right) to see how other people got along with Android ports.  I know some other people attempted it, so if you don't find a fully functional project there, hopefully it will at least get you started.  Good luck!

Lynnden42, 2017-02-05:
@SammyIAm I've looked in the forks, and from what i can make sense of, nobody has tried, or at least, nobody has tried and published it on github. I found someone on github using strings like "moppy" in their program (https://github.com/shijiameng/Moppy) if you wanna take a look. but I can't really work it out.
Would all I really need to do, (for the UI) is just work one out in a seperate project and then drop your files in and wire everything up? this'll be my first project working with android/java (i worked with c++ in the vaguest of script kiddy definitions) and i've avoided java like the plague for the longest time

Lynnden42, 2017-02-05:
@SammyIAm also thanks for responding at this ungodly hour (assuming you live in the US)

SammyIAm, 2017-02-05:
I'll try to remember to take a look at some forks later when it's a little less late (I mean, I'm on the west coast, so it's only like 11pm, but that's too late for serious thinking).

Basically though, yes, you'd just take the whole project and replace my existing ui.* classes with your own (most of the UI should be contained within the "ui" directory if I've done an okay job of separation of concerns).  It'll actually probably be easiest if you follow some sort of Android template / guide to build a simple UI first, and then import all of the Moppy java classes (except the UI) into your project and wire things up.

Make sure you've got some sort of Arduino serial communication thing worked out first as that'll probably be the most challenging (though again, probably not TOO hard and probably already solved somewhere online).

Lynnden42, 2017-02-05:
@SammyIAm so just look up something like "android to arduino serial connection" or something along those lines? (and east coast so it's late for me. I just have insomnia so i'm up until rediculous hours)

Lynnden42, 2017-02-05:
So after spending a fair chunk of last night and this morning fiddling with "https://github.com/shijiameng/Moppy/search?utf8=%E2%9C%93&q=id&type=Code" this. I can successfully say that this is in fact, NOT related to "us"

solidsnake745, 2017-02-05:
You're absolutely able to run Moppy without opening the IDE, I've been doing so for years. I should have shared this sooner so I apologize. I'm just really consumed with my family and work situations. I'm the only one who provides for my mom and my brother at the moment and that means I have to excel at my job to do so. I'll try to update the Readme with this information at some point, but until then:

Java was meant to be multi-platform (Windows, Linux, etc.) compatible and so it's more than capable of running on lots of different devices. I can't advise on anything but Windows though so I'm only going to outline what you need to achieve that. Hopefully that gives you some guidance for the Android port you want to do, but I have some additional thoughts on that which I'll outline in another post.

When you build the MoppyDesk solution, it produces a Java runtime file which is the compiled version you're looking for. You just have to install a Java runtime environment (JRE) in order to allow Windows to open and run that file. Since Sammy has updated Moppy to use nrJavaSerial, there isn't any other dependencies to consider which greatly simplifies this.

So first make sure you install a JRE from https://java.com/en/download/. To make it easy, install both 32 and 64 bit as there a number of factors that determine which version Moppy is going to try to use. I'm not going to get into those details as it's easy enough to just install both and not worry about it after that point.

Once you've done that, you're pretty much set to use Moppy by itself without the IDE. The following video runs through the process really quickly: https://youtu.be/5QDMVKNa1ig

Assuming you've already managed to get Moppy to build:

1. Open the IDE
2. Open Properties on the Moppy project
3. Navigate to the Project Folder copied from Properties
4. You'll see a 'dist' folder there - open that folder
5. The compiled JAR you're looking for should be there named the same as your project's name (default is MoppyDesk)

At that point you can run that file and that should get your standalone MoppyDesk application. I recommend making a shortcut to that file on your desktop so that if you make any changes in Netbeans to your version, you'll always be hitting the latest built version rather than have to update your copied file.

If you wanted to be able to copy just this file out to some other location (I.E. your Desktop), you'll need to take the dependencies with it or place them somewhere central so MoppyDesk can access it. That sounds really technical I know, but it's easy. The only dependency is nrJavaSerial. Take that file and copy it to the JRE's lib ext folder for both 32 and 64 bit:
[image]

That's a simple overview of what you need for Windows. My personal suggestion is to find a way to make Netbean's output something that can be run by Android and then make your UI changes to make it more mobile friendly.

solidsnake745, 2017-02-06:
I've been wanting a more mobile version of Moppy as well and have been thinking and working on this for the last year or so. I don't think porting the Java application is the way to go. Instead I've been working on porting everything to the microcontroller. I've been somewhat successful in doing this. Here is a demonstration of my progress so far: https://www.youtube.com/watch?v=VzPS0xZ5W6o

Using a Teensy 3.2, I'm able to load up MIDI files onto a SD card and play those files onto the steppers. I've implemented pin mapping as well as to free up other pins for other functionality such as push buttons and an LCD screen. With this, you should be able to entirely skip having another device as input and allow you a lot more possibilities for different kinds of setups.

I wanted to separate out the concept of MIDI channels from devices so future iterations of this library focus on everything being it's own device irrespective of channel. After this first library is done I wanted to come back and write a second one to handle pooling like the Moppy application allows for. This way you could define different strategies for different sets of devices.

That video was the first version, but I've made a few iterations since then which are much cleaner and offer more functionality among other improvements. Unfortunately, in my haste I didn't write a load test and my latest version fails horribly when playing more than one device. So now I have to go back and re-work the code which is really time consuming. Again, I don't have a lot of free time so I've had to put this on hold.

I'm providing my first version here if anyone is interested in playing around with it. I'm not going to explain too much around it or provide documentation yet because as I said there are other iterations with changes that conflict. Hate to introduce breaking changes, but if you decide to use this, know that future versions are going to change how you use it.

[Load_Test_v1.zip](https://github.com/SammyIAm/Moppy/files/753447/Load_Test_v1.zip)

Lynnden42, 2017-02-06:
@solidsnake745 Thanks for both posts (and your vids, they're great. keep up the fantastic work). This actually solved the original problem to this issue. now my issue is the android port. My issue right now is figuring out where all the files should go. Should I just dump all the files into one part of the android project? or should I place the files where'd they be in an android app? I can work with the UI as that's just making things on screen with variables that point to the program in the backend. I can work with that. I just have no idea where to place the individual files, theirselves. (an example being the src file. should I copy/paste the contents of this file into the src file in the android file structure? or should I just take the entire file structure of the moppy solution and dump it into one file in the android program?

closed 2017-02-08 by Lynnden42