samphonic/MoppyClassic #153: Cant upload code...

Failed08, 2017-02-09:
hi, can anyone tell me what is going on?
when i try to upload the code to the arduino i get this error:
Arduino: 1.8.1 (Windows 7), Board: "Arduino/Genuino Uno"

C:\Users\Tech\Desktop\Moppy-moppy-advanced\Arduino\Moppy\Moppy.ino:8:0: warning: "RESOLUTION" redefined

 #define RESOLUTION 40 //Microsecond resolution for notes

 ^

In file included from C:\Users\Tech\rekt\Moppy-moppy-advanced\Arduino\Moppy\Moppy.ino:1:0:

C:\Program Files (x86)\Arduino\libraries\Timerone/TimerOne.h:42:0: note: this is the location of the previous definition

 #define RESOLUTION 65536    // Timer1 is 16 bit

 ^

libraries\Timerone\TimerOne.cpp.o (symbol from plugin): In function `TimerOne::disablePwm(char)':

(.text+0x0): multiple definition of `TimerOne::disablePwm(char)'

sketch\TimerOne.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\Timerone\TimerOne.cpp.o (symbol from plugin): In function `TimerOne::disablePwm(char)':

(.text+0x0): multiple definition of `__vector_13'

sketch\TimerOne.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\Timerone\TimerOne.cpp.o (symbol from plugin): In function `TimerOne::disablePwm(char)':

(.text+0x0): multiple definition of `Timer1'

sketch\TimerOne.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\Timerone\TimerOne.cpp.o (symbol from plugin): In function `TimerOne::disablePwm(char)':

(.text+0x0): multiple definition of `TimerOne::setPeriod(long)'

sketch\TimerOne.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\Timerone\TimerOne.cpp.o (symbol from plugin): In function `TimerOne::disablePwm(char)':

(.text+0x0): multiple definition of `TimerOne::initialize(long)'

sketch\TimerOne.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\Timerone\TimerOne.cpp.o (symbol from plugin): In function `TimerOne::disablePwm(char)':

(.text+0x0): multiple definition of `TimerOne::setPwmDuty(char, int)'

sketch\TimerOne.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\Timerone\TimerOne.cpp.o (symbol from plugin): In function `TimerOne::disablePwm(char)':

(.text+0x0): multiple definition of `TimerOne::detachInterrupt()'

sketch\TimerOne.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\Timerone\TimerOne.cpp.o (symbol from plugin): In function `TimerOne::disablePwm(char)':

(.text+0x0): multiple definition of `TimerOne::resume()'

sketch\TimerOne.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\Timerone\TimerOne.cpp.o (symbol from plugin): In function `TimerOne::disablePwm(char)':

(.text+0x0): multiple definition of `TimerOne::pwm(char, int, long)'

sketch\TimerOne.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\Timerone\TimerOne.cpp.o (symbol from plugin): In function `TimerOne::disablePwm(char)':

(.text+0x0): multiple definition of `TimerOne::attachInterrupt(void (*)(), long)'

sketch\TimerOne.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\Timerone\TimerOne.cpp.o (symbol from plugin): In function `TimerOne::disablePwm(char)':

(.text+0x0): multiple definition of `TimerOne::start()'

sketch\TimerOne.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\Timerone\TimerOne.cpp.o (symbol from plugin): In function `TimerOne::disablePwm(char)':

(.text+0x0): multiple definition of `TimerOne::restart()'

sketch\TimerOne.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\Timerone\TimerOne.cpp.o (symbol from plugin): In function `TimerOne::disablePwm(char)':

(.text+0x0): multiple definition of `TimerOne::stop()'

sketch\TimerOne.cpp.o (symbol from plugin):(.text+0x0): first defined here

libraries\Timerone\TimerOne.cpp.o (symbol from plugin): In function `TimerOne::disablePwm(char)':

(.text+0x0): multiple definition of `TimerOne::read()'

sketch\TimerOne.cpp.o (symbol from plugin):(.text+0x0): first defined here

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino/Genuino Uno.

pls tell me whats going on and how to fix this...

SammyIAm, 2017-02-22:
You need to install the TimerOne library to your Arduino IDE.  Google is your friend for this one. :)

SammyIAm, 2017-02-22:
Oops, that's embarrassing, that's not the issue...  Hang on a minute.

SammyIAm, 2017-02-22:
Okay, two issues it looks like (thought maybe only one):

It looks like you may have two TimeOne libraries installed somehow.  It looks like one might be in a folder called "sketch" and the other is under "libraries".  Try tracking these both down and move / delete one of them.

*Hopefully* that solves your issue (it should at least get rid of most of those errors).  If the `RESOLUTION` error is still sticking around, open up Moppy.ino and replace any instances of `RESOLUTION` you find with something like `MOPPY_RESOLUTION` to avoid the ambiguity.

Let me know if those work (sorry for the quick dismissal before)

Goemansj, 2017-03-29:
What version of TimerOne did u download?

SammyIAm, 2017-06-04:
I used the one linked from the README:
https://github.com/SammyIAm/Moppy