samphonic/MoppyClassic #155: fatal error: TimerOne.h: No such file or directory

6829, 2017-03-12:
Im Not sure why im getting a error since i have TimerOne installed in the right directory.
Here is the error code.
C:\Users\KING\Documents\Arduino\Moppy\Moppy\Moppy.ino:1:22: fatal error: TimerOne.h: No such file or directory

 #include <TimerOne.h>

                      ^

compilation terminated.

exit status 1
Error compiling for board Arduino/Genuino Uno.
Invalid library found in C:\Users\KING\Documents\Arduino\libraries\examples: C:\Users\KING\Documents\Arduino\libraries\examples
Invalid library found in C:\Users\KING\Documents\Arduino\libraries\examples: C:\Users\KING\Documents\Arduino\libraries\examples

Many Thanks, Hopefully will get it sorted

AmEv7Fam, 2017-03-13:
http://playground.arduino.cc/Code/Timer1

> Download -> TimerOne Google Code download

> To install, simply unzip and put the files in Arduino/hardware/libraries/Timer1/

6829, 2017-03-13:
Hi 
I have tried that and still comes up with an error!
Also i don't have a folder exactly like Arduino/hardware/libraries/Timer1/
Im quite confused :/

Arduino: 1.8.1 (Windows 10), Board: "Arduino/Genuino Uno"

C:\Users\KING\Documents\Arduino\Moppy\Moppy\Moppy.ino:1:22: fatal error: TimerOne.h: No such file or directory

 #include <TimerOne.h>

                      ^

compilation terminated.

exit status 1
Error compiling for board Arduino/Genuino Uno.

Thanks :)

6829, 2017-03-13:
Im putting timer One in here:
C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries
I have also tried putting it in:
C:\Users\KING\Documents\Arduino\libraries
And it still comes up with the same thing!

Thanks

6829, 2017-03-13:
Its fine now , i fixed it :)
Thanks for the help

closed 2017-03-13 by 6829

poopoo1993, 2017-09-20:
Would you plz tell me that how did you fix it

DjoleBuk, 2017-11-15:
Hello, yeah I have the same problem, do you remeber how did you fix it ?

thanks

6829, 2017-11-16:
Yes, well you go into this plugin search thing then put in timerone and
download it then it works!

SammyIAm, 2017-11-21:
This is more of an Arduino IDE issue question.  You can find the TimerOne library here: https://playground.arduino.cc/Code/Timer1

Or in newer versions of the IDE, I believe it's available from the built-in library manager.