samphonic/MoppyClassic #156: Netbeans

6829, 2017-03-13:
Hi
I have Ardunio all set up and working but i am not sure how to set up the Netbean bit as there is not a guide.

Hopefully will get this sorted out :)
Thanks

SammyIAm, 2017-06-04:
Hi there, I know this response is rather late (but better than never, I guess...).  You can learn about installing / setting up NetBeans from NetBean's website.  After that, you can just open up the MoppyDesk project in NetBeans and press run.  Two things you may end up needing to check:
* There may be some sort of NetBeans UI library required.  If you're seeing an error about missing some sort of Application class, you can use Google to figure out the latest NetBeans plugin that you'll need.
* Depending on which OS you're using, you may need to adjust the serial drivers to match.  There's not really a guide on this one because it varies a lot system-to-system, but again Googling for setting up serial drivers for Java in your OS should be fine (Moppy isn't doing anything especially unique in this regard)

closed 2017-06-04 by SammyIAm

6829, 2017-06-04:
Thanks, i have sorted it out now anyway :)

6829, 2017-06-09:
Cheers for the info, helped a lot :)
Also thanks for the tips for the moppy desk
Alex