samphonic/MoppyClassic #159: Connecting Error.

GlorifyedToasterWithLegs, 2017-04-02:
When my Moppy Control App Froze and stopped working I closed out of it by using task manager but when I tried to load the program up again, when I the Connect Button it gave me the error reading "java.lang.NullPointerException"  Help Would Be Greatly Appreciated!

GlorifyedToasterWithLegs, 2017-04-03:
UPDATE!!!
I just restarted my computer and now I have no COM Ports to select from (only com port 1)

SammyIAm, 2017-06-04:
Uh oh, that's no good.  You may want to disconnect your Arduino and plug it back into a different USB port (sometimes that helped for me for some reason).  The Arduino-COM connection seems to get grumpy sometimes if it's not closed properly.